dotintent/react-native-ble-plx #688: Scanning BLE cannot get the ADV payload, manufactureData: null

RadioOperator, 2020-06-04:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior
Using react-native-ble-manager LIB, we got:

{"advertising": {"isConnectable": true, 
"manufacturerData": {"CDVType": "ArrayBuffer", 
"bytes": [Array], 
"data": "Fxbk/vvX5mCIAEoPAG/2+GQAAG8AAIABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="},
"serviceData": {"fee4": [Object]}, 
"serviceUUIDs": [],
"txPowerLevel": -2147483648},
"id": "D7:FB:D0:83:09:73",
"name": "SC962_D7FB",
"rssi": -49}

the "data" (string) is exactly right what we wanted.

## Current Behavior
the same device, your blt-lpx lib, we got:

{"_manager": {"_activePromises": [Object], 
"_activeSubscriptions": [Object], 
"_errorCodesToMessagesMapping": [Object], 
"_eventEmitter": [NativeEventEmitter], 
"_scanEventSubscription": [EmitterSubscription], 
"_uniqueId": 6}, 
"id": "D7:FB:D0:83:09:73", 
"isConnectable": null, 
"localName": null, 
"manufacturerData": null, 
"mtu": 23, 
"name": "SC962_D7FB", 
"overflowServiceUUIDs": null, 
"rssi": -54, 
"serviceData": {"0000fee4-0000-1000-8000-00805f9b34fb": "+9faXYgASg8Ab/b4ZAAAmAEAgAE="}, 
"serviceUUIDs": null, 
"solicitedServiceUUIDs": null, 
"txPowerLevel": null}

the "manufacturerData": null, in above.

Could you improve on this?
or tell me where is the handler capturing the ADV payload to manufacturerData.

Thanks

Cierpliwy, 2020-06-15:
Can you show snipped of the code, which prints above logs?

RadioOperator, 2020-06-15:
@Cierpliwy hi, 
The BLE manager project is [https://github.com/arricor/react-native-ble-poc](https://github.com/arricor/react-native-ble-poc)
and your plx project is [https://github.com/joernb/react-native-ble-scanner](https://github.com/joernb/react-native-ble-scanner)

The original scanner results in JSON format, I added CR/LF in above list to show it clearly.

Here show a Screenshot from ble-manager project. In the picture, the 1st device is the decoded Base64 payload data (same BLE device).

[image]

kncs, 2020-08-27:
@RadioOperator  did you find why manufacturerData was null ? Or where to get ADV payload ?

RadioOperator, 2020-08-28:
@kncs Hi, no, do not know why.
Maybe this Lib cannot get the right Payload for a non-standard data likes my device, the data starts 17 16 E4 FE.......

tony-dinh-spi, 2021-01-05:
I am having the same problem. Has this been resolved or figured out?

RadioOperator, 2021-01-07:
@tony-dinh-spi  Hi, still not resolved. I have to use "react-native-ble-manager".