dotintent/react-native-ble-plx #689: Connect to a device using its mac address

tolik85, 2020-06-04:
Hello!

Can I connect to a device using its mac address only, like in native implementation of this library?
I tried to find my devices using BleManager.devices('aa:aa:aa:aa:aa:aa'), but it's returning an empty array.
The goal is to connect to my specific device without making a BLE scan.

Thank you!

dariuszseweryn, 2020-06-05:
On iOS there is no possibility to get a MAC address of a device. On Android it works if the MAC address is static.

closed 2020-06-05 by dariuszseweryn

RRMoelker, 2020-06-26:
Is is required to scan before connecting to the device using its unique id?

And if it is required, is that a hard limitation by the underlying OS or something that can be improved in the library?

p.s. This question seem in line with original issue. If you disagree I'll happily create a new ticket.

dariuszseweryn, 2020-06-30:
> Is is required to scan before connecting to the device using its unique id?

On iOS it is not required. On Android it is not required if the device is bonded or has been recently connected (and there was no bluetooth adapter powercycle in between) — otherwise YMMV — it is OS bug/limitation. Unfortunately there is no bulletproof way to fix it in library so it is always better to scan on Android before connecting if a not bonded device is used.

Datagrdn, 2020-10-21:
@dariuszseweryn I see you mention that you can connect to a device with MAC address alone on Android. Can you share how? In the documentation I'm just seeing use of UUID.

Thank you!