dotintent/react-native-ble-plx #692: onDisconnected even is not firing immediately once connection lost(If switch off the ble device)

priyanka-Sadh, 2020-06-06:
I am using as version below.
react-native": "0.62.2",
"react-native-ble-plx": "^1.1.1"

I want know and fix ->there should not be a delay in calling `onDisconnected` method.
Scenario : In mobile app scanning, gets connected with device, Once I switch off ble device, onDisconnected method take 4-5 seconds to call. I want it should be immediately disconnect. Is it possible ?

dariuszseweryn, 2020-06-06:
Yes and no. Yes if the peripheral you turn off will send "connection end" signal before switching off. No if it won't — then a Bluetooth Supervision Timeout needs to kick in which is platform dependant and there is no direct control over it from the app side.

closed 2020-06-06 by dariuszseweryn

priyanka-Sadh, 2020-06-06:
Thanks for comment