dotintent/react-native-ble-plx #693: Endless scanning sometimes.

priyanka-Sadh, 2020-06-06:
Using following version
"react-native": "0.62.2",
"react-native-ble-plx": "^1.1.1",
I want to know reason of below scenario: 

When Mobile app is in connected status with ble device, 
and I turn off the device app goes to disconnected mode and re-scan and then connect. (I repeat this)
It happens regular 2-3 time successfully, after this sometime app goes to endless scanning (Scanning...) mode so in this case I have to kill and re-launch the app.

And sometime  (after 2-3 time) it continuously says `BleError : can not start scanning operation`
so in this case I have to kill and re-launch the app or need to again turn-off or turn on the device.

My code below
`this.manager.startDeviceScan(null,{ScanMode : "Balanced" }, (error, device) => {
    console.log("Scanning...");
    if (null) {
        console.log('null');
        return;
    }
    if (error) {
        console.log("Error in scan=> "+error)
        this.scanAndConnect()
        return
    }
    if(device.name == 'mydevice'){ 
        this.manager.connectToDevice(device.id, {autoConnect:true}).then((device) => {
            (async () => {
                            //listener for disconnection
                device.onDisconnected((error, disconnectedDevice) => {
                    console.log('Disconnected ', disconnectedDevice.name);
                    this.scanAndConnect(); // to start scanning again
                });
            })();    
        });
    }
})`

may I know the reason or any mistake if I made here in code ?

dariuszseweryn, 2020-06-06:
You do not stop the scan it seems?

priyanka-Sadh, 2020-06-06:
I do just after once mydevice got found. Sorry for above code mistake.
`this.manager.startDeviceScan(null,{ScanMode : "Balanced" }, (error, device) => { console.log("Scanning..."); if (null) { console.log('null'); return; } if (error) { console.log("Error in scan=> "+error) this.scanAndConnect() return } if(device.name == 'mydevice'){ this.manager.stopDeviceScan();this.manager.connectToDevice(device.id, {autoConnect:true}).then((device) => { (async () => { //listener for disconnection device.onDisconnected((error, disconnectedDevice) => { console.log('Disconnected ', disconnectedDevice.name); this.scanAndConnect(); // to start scanning again }); })(); }); } })`

priyanka-Sadh, 2020-06-06:
here is the error {console.log(JSON.stringify(error));}
`[Sat Jun 06 2020 21:26:23.869]  LOG      {"message":"Cannot start scanning operation","errorCode":600,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Undocumented scan throttle (code 2147483646), suggested retry date is Sat Jun 06 21:26:21 GMT+05:30 2020","name":"BleError","line":26121,"column":28,"sourceURL":"http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"}
`

dariuszseweryn, 2020-06-06:
You scan too often. Android allows only 5 scans in every 30 seconds window.

priyanka-Sadh, 2020-06-06:
same is for ios ?

dariuszseweryn, 2020-06-06:
No

priyanka-Sadh, 2020-06-06:
hmm, I checked with stop watch, continuous =>scan,connect,disconnect (Repeat) work for 2 time only, in 3rd round it goes endless scanning-> {"message":"Cannot start scanning operation"} and this error.

Check with stopwatch 3rd round was under 30seconds window.

priyanka-Sadh, 2020-06-06:
Full working file
`import React,{Component} from 'react';
import {BleManager} from "react-native-ble-plx";
import {PermissionsAndroid, Platform} from 'react-native';

export default class Ble_test extends Component {
    constructor() {
        super();
        if (Platform.OS === 'android') {
            PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
                .then(granted => {
                    if (granted !== 'granted') {
                        console.log('BLUETOOTH denied')
                    }
                    else {
                        this.initializeManager();
                    }
                })
        } else {
            this.initializeManager();
        }

    }

    initializeManager() {
        this.manager = new BleManager({
            restoreStateIdentifier: 'testBleBackgroundMode',
            restoreStateFunction: bleRestoredState => {
            }
        });
        const subscription = this.manager.onStateChange((state) => {
            if (state === 'PoweredOn') {
                this.scanAndConnect();
            }
        }, true);
    }

    scanAndConnect() {
        this.manager.startDeviceScan(null, null, (error, device) => {
            console.log("scanning");
            console.log(device);
            
            if (error) {
                console.log(JSON.stringify(error));
                this.scanAndConnect()
                return
            }
            
            if(device.name == 'TAPP_noljal'){
                console.log("found ", device.name);
                this.manager.stopDeviceScan();
                this.manager.connectToDevice(device.id, {autoConnect:true}).then((device) => {
                        (async () => {
                            console.log("connected to ",device.name);
                            //listener for disconnection
                            device.onDisconnected((error, disconnectedDevice) => {
                                console.log('Disconnected ', disconnectedDevice.name);
                                this.scanAndConnect();
                            });
                            
                        })();
                    })
         }
        });
    }

    render() {
        return null;
    }
}`

dariuszseweryn, 2020-06-06:
Correct me if I am wrong but returning an error is not an endless scan? Is it? It is an error while starting a scan?

priyanka-Sadh, 2020-06-06:
Yes,
So  I am calling `this.scanAndConnect()` again if error comes.

dariuszseweryn, 2020-06-06:
Mhm... And what do you recon then?

priyanka-Sadh, 2020-06-06:
Yes, I want reconnect then with same device.
Because user who is using BLE device can regular switch off and switch on the device. This can be a scenario.

dariuszseweryn, 2020-06-06:
There is nothing you can do. There may be only that many scans during a specific time frame. You have to deal with it

priyanka-Sadh, 2020-06-06:
Ahh, Okay, 
Well Thanks for responding so fast. I am closing it now, If in case further scanning query I will reopen it. 
I don't know if i am referring wrong here for replying on https://github.com/Polidea/react-native-ble-plx/issues/669 please

closed 2020-06-06 by priyanka-Sadh

sunilloharshoreline1, 2021-02-12:
Any solution found? I m facing the same issue. After connecting and disconnecting the Peripheral two times, it doesn't start the scan again.

chixor, 2021-11-15:
FWIW the quick start instructions pretty much guarantee a new developer will encounter this throttle error. It would be nice if the instructions guided 'best practice' instead of 'rough and ready'.