dotintent/react-native-ble-plx #695: Bug Report

manojrai84, 2020-06-08:
Platform: Android

using tge ble libraray its working fine for iOS in android not able to scan it throws null if i console the device.name using 0.61 react?

Please help

dariuszseweryn, 2020-06-08:
Hello,

I do not understand what do you mean by "it throws null if console the device.name". I can only imagine that you refer to "method called on null object reference" — if that is the case the `device` is not available and therefore the callback was called with an `error`.

manojrai84, 2020-06-09:
Hi Dariusz,
Thanks for your quick response, there is no error it’s throwing but if I just print console.log(device),but if I keep the device for scan for sometime like around 2-3 min than sometimes it’s showing few device only .
PFA

dariuszseweryn, 2020-06-09:
In terms of how many devices you would get on prolonged scan I cannot give you estimates as it may vary greatly. 
I am not sure what I can help you with.

manojrai84, 2020-06-09:
I am sitting next to my MAC and it not able to scan and it’s not showing the list on the screen also in the android and same code works fine for the iOS . So do you mind checking once installing the code as it’s not working on Samsung S8 device

dariuszseweryn, 2020-06-09:
> same code

What code?

manojrai84, 2020-06-10:
Hi Polidea/react-native-ble-plx,

Error: Unable to resolve module
`react-native-permissions/lib/typescript/constants` from
`actions/index.js`: react-native-permissions/lib/typescript/constants could
not be found within the project.

suddenly this error started coming after doing these 3 steps still see the
same issue i havent did anything new , can you please :

1. Clear watchman watches: watchman watch-del-all

 2. Delete node_modules: rm -rf node_modules and run yarn install

 3. Reset Metro's cache: yarn start --reset-cache

 4. Remove the cache: rm -rf /tmp/metro-*

dariuszseweryn, 2020-06-29:
Sorry for such a delay.

You seem to have issues with react-native and not with the library... There are plenty of those already resolved. Try googling.

As for the scanning. Check scan options for low latency scan.

closed 2020-06-29 by dariuszseweryn