dotintent/react-native-ble-plx #696: [Bug Report] It cannot get ble indicate info rightly

shangzuoyan, 2020-06-09:
## Expected Behavior

I supposed that it would get the ble indicate info
for example:
"00002a35-0000-1000-8000-00805f9b34fb": [Object],
Blood Pressure Measurement
org.bluetooth.characteristic.blood_pressure_measurement

## Current Behavior

It missed the ble indicate everytime.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Connect to a a blood pressure ble device.
2. Perform a check
3. No ble indicate result

Thanks
Roy

dariuszseweryn, 2020-06-09:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/react-native-ble-plx/blob/master/.github/ISSUE_TEMPLATE/bug_report.md)?

Best Regards

dariuszseweryn, 2020-06-29:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

closed 2020-06-29 by dariuszseweryn

shangzuoyan, 2020-07-07:
Thanks a lot.
This issue was fixed. I register a callback for that notification characteristic.

dariuszseweryn, 2020-07-07:
Glad to hear that. 👍  More often than not these kind "bug reports" are questions on how to properly use BLE. Cheers!