dotintent/react-native-ble-plx #697: Question : OnDisconnect not fixed yet ?

josef256, 2020-06-11:
the onDisconnect Listener is fixed ? cause the branch which was supposed to fix it was merged but after updating the library the issue still persist.

tested on Android.
#679

iDorosh, 2020-06-17:
I am also still having the same issue on android.

iDorosh, 2020-06-17:
Curious about this too. Disconnect listener not working on android

Barbiero, 2020-06-18:
while #679 was definitely merged, the patch itself was not released on npm nor github for that matter - latest is still 2.0.0.

As a workaround, until it releases, you can install the package by the most recent commit:

```
npm install https://github.com/Polidea/react-native-ble-plx.git#387e4f080f953b0476faa9c06cdd448aa7a1a1b4
```

iDorosh, 2020-06-18:
Ok perfect thanks for the info! Works great

dariuszseweryn, 2020-06-30:
Just released `2.0.1`

closed 2020-06-30 by dariuszseweryn