dotintent/react-native-ble-plx #698: Question

hern12, 2020-06-11:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ yes ] I am running the latest version
- [ yes ] I checked the documentation and found no answer
- [ yes ] I checked to make sure that this issue has not already been filed
- [ yes ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.
can I find another mobile phone by using this library

Daybayzayd, 2020-06-15:
Maybe duplicate of this question ?

https://github.com/Polidea/react-native-ble-plx/issues/644

dariuszseweryn, 2020-06-15:
Duplicate

closed 2020-06-15 by dariuszseweryn