dotintent/react-native-ble-plx #699: startDeviceScan listener never called iOS background

RRMoelker, 2020-06-25:
## Expected Behavior

I'm calling `manager.startDeviceScan(['myservice-uuid'], listener)` in code that is run when a Apple Push Notification is received. I expect the callback to be called (and the device to be found) even while processing is happening in background. But it seems the device search does not operate in the background.

## Current Behavior

The callback is called when app is in foreground.
The callback is **never** called when `startDeviceSearch` function is called in the background.

The problem occurs when handling a push message or when running in a background timer.

Tried to no avail:
* Setting `restoreStateIdentifier` and `restoreStateFunction`
* Providing the UUID to search for

### Steps to Reproduce

For me the problem is that I can't scan when responding to a push message. But the same problem also occurs when using a background timer. Therefor the easiest way to reproduce is to use a timer.

I've modified this [example project](https://github.com/PolideaPlayground/BleInTheBackground-iOS) to showcase the bug: [modified fork](https://github.com/RRMoelker/BleInTheBackground-iOS)

Steps
1. Clone https://github.com/PolideaPlayground/BleInTheBackground-iOS
2. Set "const serviceUUID" to your device id.
3. Build & run
4. Press "**Scan FG**". Notice that the listener log is shown in Metro console. (**expected**)
5. Press "**Scan BG**"
6. Close the app to background
7. See that no listener callback console statement is shown in Metro console. (**unexpected**)

### Context

* Library version: example repo is: **1.1.1**, but also occurs on **2.0.0**
* Platform: iOS
* Platform logs (logcat/XCode): 

"Scan FG" pressed:
https://pastebin.com/BZPbB7aj

"Scan BG" pressed:
https://pastebin.com/EV1d3iyJ

* JS logs:

"Scan FG" pressed:
```
LOG  1593090492584: Waiting for PoweredOn state...
 LOG  1593090492633: BLE state: PoweredOn
 LOG  1593090492675: Scanning for device...
 LOG  starting device scan
 LOG  >>> device scan listener called!
```

"Scan BG" pressed:
```
 LOG  1593090523705: Waiting for PoweredOn state...
 LOG  1593090523753: BLE state: PoweredOn
 LOG  1593090523790: Starting bg scan
 LOG  1593090524992: App state: inactive
 LOG  1593090525588: App state: background
 LOG  1593090527089: Background timer fire
 LOG  starting device scan
<nothing>
```

* Contents of the `package.json` file:
see repo
* Formatted code sample or link to a repository:
https://github.com/PolideaPlayground/BleInTheBackground-iOS

```js
    console.log('starting device scan');
    manager.startDeviceScan(
      [serviceUUID],
      {allowDuplicates: true},
      (error, scannedDevice) => {
        console.log('>>> device scan listener called!'); // not called when started from background process
        ...
    }
```

RRMoelker, 2020-06-26:
Putting the ticket on hold myself.

The problem is probably that iOS is doing a "passive" scan instead of an active one. That means it requires the service UUID to be in the peripheral advertisement packet. For now it seems that is the problem for me is that is not advertised.

It's probably a good idea to add the passive scan wisdom to: https://github.com/Polidea/react-native-ble-plx/wiki/Background-mode-(iOS)

Further reading on the issue: https://medium.com/@cbartel/ios-scan-and-connect-to-a-ble-peripheral-in-the-background-731f960d520d

closed 2020-06-26 by RRMoelker

joaomdmoura, 2021-06-14:
@RRMoelker did you were able to fix this properly updating the characteristic  string to include the UUID? I tried to do it on my sidee, but am still experiencing issues

RRMoelker, 2021-06-15:
Hello @joaomdmoura, it has been a while since I encountered this problem. From what I recall we ended up updating the firmware so the service UUID was send out from our Bluetooth device.

joaomdmoura, 2021-06-15:
Got it, makes sense! thanks, super appreciate it!