dotintent/react-native-ble-plx #700: Monitoring

ad-boschung, 2020-06-25:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

When I call the `remove()` function on the monitor characteristic `Subscription`, 

## Current Behavior

I start monitoring my characteristic and assign the `Subscription` received. Everything is fine.
Then I call the `remove()` function on the `Subscription` when I want to stop monitoring the characteristic.

1) At this time, I receive a `BleError : Operation was cancelled` in the subscription error callback.
I don't understand why I receive an error if I remove() the subscription. It's not an error.

2) After that, I need to monitor my characteristic again, so I start monitoring again. And at this time, I have a `Characteristic X notify change failed for device ? and service Y`.
But it was working well the first time I called the monitor function... I don't understand why I receive this error.
I noticed that if I disconnect from the device, and reconnect agin I can monitor without issues.

See examples below.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.0

```js
  function startMonitoring() {
    console.log('LIB : Start fifo characteristic monitoring...');
    this._monitorSubscription = this._fifoCharacteristic.monitor(
      (error, char) => {
        if (error) {
          console.log(error);
        }
        if (!char || !char.value) {
          console.log(char.value);
        }
      },
    );
  }

  function stopMonitoring() {
    this._monitorSubscription.remove();
  }
```

dariuszseweryn, 2020-06-29:
Again, you did not conform to the issue template. I have checked this scenario and it works correctly for me:
```
2020-06-29 14:59:32.173 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: true
2020-06-29 14:59:32.178 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(92869694)
2020-06-29 14:59:32.181 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(92869694)
2020-06-29 14:59:32.290 1345-1358/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:32.293 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(92869694) in 114 ms
2020-06-29 14:59:34.336 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 0
2020-06-29 14:59:34.392 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 1
2020-06-29 14:59:34.471 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: false
2020-06-29 14:59:34.472 1345-1484/com.reactnativebleplxplayground I/ReactNativeJS: 'monitor error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError","line":26115,"column":28,"sourceURL":"http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"}'
2020-06-29 14:59:34.475 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(121702638)
2020-06-29 14:59:34.476 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(121702638)
2020-06-29 14:59:34.582 1345-1358/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:34.585 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(121702638) in 109 ms
2020-06-29 14:59:43.101 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 0
2020-06-29 14:59:43.225 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 1
2020-06-29 14:59:43.268 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: true
2020-06-29 14:59:43.276 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(153297124)
2020-06-29 14:59:43.277 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(153297124)
2020-06-29 14:59:43.358 1345-1399/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:43.360 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(153297124) in 83 ms
2020-06-29 14:59:48.460 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 0
2020-06-29 14:59:48.617 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 1
2020-06-29 14:59:48.633 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: false
2020-06-29 14:59:48.637 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(16943892)
2020-06-29 14:59:48.640 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(16943892)
2020-06-29 14:59:48.693 1345-1484/com.reactnativebleplxplayground I/ReactNativeJS: 'monitor error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError","line":26115,"column":28,"sourceURL":"http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"}'
2020-06-29 14:59:48.769 1345-1358/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:48.771 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(16943892) in 131 ms
2020-06-29 14:59:50.898 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 0
2020-06-29 14:59:51.011 1345-1345/com.reactnativebleplxplayground D/ViewRootImpl@2291129[MainActivity]: ViewPostImeInputStage processPointer 1
2020-06-29 14:59:51.071 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: true
2020-06-29 14:59:51.079 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(220331194)
[... cut ...]
2020-06-29 14:59:53.032 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(122882154)
2020-06-29 14:59:53.034 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(122882154)
2020-06-29 14:59:53.107 1345-1357/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:53.110 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(122882154) in 76 ms
```

dariuszseweryn, 2020-06-29:
I have checked on iOS and it also seems to work just fine:
```
2020-06-29 17:16:45.791 [info][tid:com.facebook.react.JavaScript] connected
[RxBLEKit|DEBG|15:16:49.366]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: true,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:16:49.434]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: true,
error: nil)
[RxBLEKit|DEBG|15:16:52.982]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: false,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:16:53.064]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: false,
error: nil)
[RxBLEKit|DEBG|15:18:06.855]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: true,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:18:06.925]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: true,
error: nil)
[RxBLEKit|DEBG|15:18:10.274]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: false,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:18:10.345]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: false,
error: nil)
[RxBLEKit|DEBG|15:18:13.509]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: true,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:18:13.586]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: true,
error: nil)
[RxBLEKit|DEBG|15:18:14.402]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) setNotifyValue(
enabled: false,
for: Characteristic(uuid: 0002, id: 10760138944))
[RxBLEKit|DEBG|15:18:14.455]: Peripheral(uuid: D735B5D2-E6B4-DF67-5AA3-5F0D5DBEE6BB, name: Optional("Test")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 0002, id: 10760138944), isNotifying: false,
error: nil)
```

ad-boschung, 2020-08-10:
Why did I not conform to the issue template ??

1. You have the same behaviour than I have. You have the `BleError : Operation was cancelled` too. But it is not an error. Why do we receive an error when we just remove the subscription ? It's not an error I guess.
2. Okay it seems to work on your side, although it is not on my side...

closed 2020-10-17 by stale[bot]

powext, 2021-02-03:
any updates? i'm facing the same problem

Akshat7321, 2021-04-06:
same here, I am also facing same issues on cancelling subscription of some stream.

makzee, 2021-05-10:
I'm facing the same issue.

tymoxx, 2022-05-11:
I confirm that `monitorSubscription.remove()` works fine on iOS but on Android it gives an error **[BleError: Operation was cancelled]**

chidaihuashi, 2023-08-24:
I also found this issue in version 2.0.3. It seems that this problem has not been fixed.

Andrrew94, 2024-06-27:
Still having this issue on version 3.2.0, any workarounds ?

paddymccallan-izon, 2024-08-30:
Same issue here on android