dotintent/react-native-ble-plx #701: Bug Report

jgcook935, 2020-06-27:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Characteristics on a device should have the same id after reconnecting on iOS.

## Current Behavior

On iOS when I connect to a device, disconnect, reconnect, ble-plx will throw a CharacteristicNotFound BLEError stating the characteristic with id: {id} cannot be found. I do see this characteristic in the console when we run monitor, but the ids are different. This same flow works on Android. In the documentation it is mentioned to make sure you run `manager.discoverAllServicesAndCharacteristicsForDevice`, but we do when we connect. 

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

I'm not sure this will be a 1 : 1, but the steps in my situation are
1. boot the app
2. connect to device
3. disconnect/hot reload
4. observe that a control point characteristic exists, but can't be written to on iOS

### Context

* Library version: 2.0.0
* Platform: iOS 13.5.1
* JS logs:
[image]

I'm not allowed to share too much code, but if anyone has a specific question I can likely answer with smaller snippets. Thank you for any input provided.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

jgcook935, 2023-10-10:
@abursztynowska Fantastic. Thank you for the follow up.

closed 2023-11-02 by abursztynowska