dotintent/react-native-ble-plx #703: Bug Report: Ios unable to find all nearby devices

d3ullist, 2020-06-29:
- [ ] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

When searching for devices, I expect to find the 4 nearby devices on both IOS and Android

## Current Behavior

On android, all 4 devices are found as expected.
On IOS only 2 of the 4 devices can be found, the ones found are always the same

### Steps to Reproduce

Grab android phone (samsung galaxy s8), and validate all 4 devices are findable in nrf connect.
Boot up the app, scan for devices, receive a list of 4 found devices. 

Grab ios phone (iphone 8), and validate all 4 devices are findable in nrf connect.
Boot up app, scan for devices, receive a list of 2 out of 4 devices.

### Context

* Library version: ^1.1.1
* React-native version: 0.59.9
* Platform: iOS, Iphone8

 [x] Validated that the devices have the correct tx, rx and service id's
 [x] Validated that the devices match the filtered name list
 [x] Validated no error message are send on #onScan

```js
class BLE {
  bleManager = new BleManager();

  devices = [];

  deviceName = 'GT ';

  searchDuration = 1280 * 10;

  stateUpdateDuration = 500;

  stateUpdateAttempts = 0;

  maxStateUpdateAttempts = 10;

  txID = '6E400003-B5A3-F393-E0A9-E50E24DCCA9E';

  rxID = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E';

  serviceId = '6E400001-B5A3-F393-E0A9-E50E24DCCA9E';

  startScan = () =>
    this.bleManager.startDeviceScan(
      [this.serviceId],
      {
        // IOS Only option
        allowDuplicates: false,
      },
      this.#onScan,
    );

  stopScan = () => this.bleManager.stopDeviceScan();

  getState = () => this.bleManager.state();

   #onScan = (error: ?Error, scannedDevice: ?Device) => {
    if (error) {
      console.log({ error });
      return;
    }

    const exists = find(this.devices, ['id', scannedDevice.id]);
    const existingName = find(this.devices, ['name', scannedDevice.name]);

    if (
      !exists &&
      !existingName &&
      scannedDevice.name &&
      scannedDevice.name.startsWith(this.deviceName) &&
      scannedDevice.name.toUpperCase() !== 'GT COM'
    ) {
      this.devices.push(scannedDevice);
    }
  };
}
```

dariuszseweryn, 2020-06-29:
I am afraid that you have not attached the video.

d3ullist, 2020-06-29:
> 
> 
> I am afraid that you have not attached the video.

It should be there now

dariuszseweryn, 2020-06-29:
Could you add `console.log(scannedDevice)` before the `if` and show the logs? Native logs could also be valueable

d3ullist, 2020-06-30:
[Logs.txt](https://github.com/Polidea/react-native-ble-plx/files/4850980/Logs.txt)
[image]

dariuszseweryn, 2020-06-30:
Could you show native logs as well? (`manager.setLogLevel(LogLevel.Verbose);`)

d3ullist, 2020-06-30:
[Logs-2.txt](https://github.com/Polidea/react-native-ble-plx/files/4852036/Logs-2.txt)

dariuszseweryn, 2020-06-30:
It seems to correctly return all that native part receives. You could try remove the native filtering for service UUID and check if it changes anything. If so post new logs please. If not compare the peripherals that are shown and not shown in nRF — they must be different in some way as it seems that the library behaves in a consistent way.

closed 2020-07-01 by d3ullist

dariuszseweryn, 2020-07-01:
@d3ullist Could you add more info? Where there any differences between those devices?

d3ullist, 2020-07-01:
The reproduction on my end (the one where the logs where coming from) seems to have not been a proper reproduction of the issue at hand. After a device reboot all 3 where properly being picked up again by ble-plx without an issue. 

The original issues stemming the bug report seem to have been caused by caching of old device names, or something along those lines.
Several weeks ago the naming scheme changed from a static GT Com to GT XXXXXXXX, and while nrf connect did manage to properly detect them under the new name. The app with ble-plx still seemed to detect the old GT Com name.
Using a python script with another bluetooth library we detected that on this end it also still detected the old GT Com name as well. After a power on/off like reset on those devices, they became available through ble-plx on the ios side as well.

dariuszseweryn, 2020-07-01:
Glad to hear that the issue is solved.

This however does not explain why your additional devices did not appear in the logs as there was only a native filter using the service UUID (nothing about name filtering). I could imagine this to be a problem if the service UUID appeared only in Scan Response and not in the Advertisement packet.

d3ullist, 2020-07-01:
@dariuszseweryn On the android side we currently do not have the service UUID filter implemented. Considering they did show up in there, and not in IOS I would also assume something along those lines.