dotintent/react-native-ble-plx #706: I can't get the listener in monitorCharacteristicForDevice to work

williamgoodhew, 2020-07-02:
I am trying to read a value from my Bluetooth low energy device.

Here is the code for the function.

```
function onConnect(){

    const myData = '0x06010101'
    const dataMostSignificantByte = (myData >> 8) & 0xFF;
    const dataLeastSignificantByte = myData & 0xFF;

    const dataByteArrayInBigEndian = [dataMostSignificantByte, dataLeastSignificantByte];
    const dataInBase64 = Buffer.from(dataByteArrayInBigEndian).toString('base64');
    console.log('dataInBase64: ', dataInBase64)

    manager.connectToDevice(scooterId).then((device) => {
    console.log("Connected");

    manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
    .then(() => { 
      console.log('discovered');
      manager.characteristicsForDevice(scooterId, '50000001-0000-1000-8000-00805F9B34FB')
    .then(() => { 
        console.log('writing');
        return manager.writeCharacteristicWithResponseForDevice(
          scooterId, 
          '50000001-0000-1000-8000-00805F9B34FB',
          '50000003-0000-1000-8000-00805F9B34FB', 
          dataInBase64
        )
    })
    .then(() => {
        console.log('monitoring');
          manager.monitorCharacteristicForDevice(
            scooterId, 
            '50000001-0000-1000-8000-00805F9B34FB',
            '50000002-0000-1000-8000-00805F9B34FB', 
            (err, characteristic) => {
              if (err) {
                console.log('error is here: ', err)
                return err;
              }
              console.log('monitor');
              console.log(characteristic.value);
              return;
            })
      })     
      })
    })
}
```

Here are the native iOS logs

```
2020-07-02 09:43:23.821 [info][tid:com.facebook.react.JavaScript] 'dataInBase64: ', 'AQE='
[RxBLEKit|DEBG|08:43:23.827]: CentralManager(10795156096) retrievePeripherals(
withIdentifiers: [DCC17464-FEFC-DB10-D46A-2252F62AD9E4])
[RxBLEKit|DEBG|08:43:23.831]: CentralManager(10795156096) connect(
peripheral: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")),
options: nil)
2020-07-02 09:43:23.839 [info][tid:com.facebook.react.JavaScript] Connected
[RxBLEKit|DEBG|08:43:23.842]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverServices(
serviceUUIDs: nil)
[RxBLEKit|DEBG|08:43:23.845]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverServices(services
: Optional("[Service(uuid: 50000001-0000-1000-8000-00805F9B34FB, id: 10795158144), Service(uuid: FE59, id: 10795121152)]"),
error: nil)
[RxBLEKit|DEBG|08:43:23.847]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverCharacteristics(
characteristicUUIDs: nil,
for: Service(uuid: 50000001-0000-1000-8000-00805F9B34FB, id: 10795158144))
[RxBLEKit|DEBG|08:43:23.848]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverCharacteristics(
characteristicUUIDs: nil,
for: Service(uuid: FE59, id: 10795121152))
[RxBLEKit|DEBG|08:43:23.851]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverCharacteristicsFor(for:
Service(uuid: 50000001-0000-1000-8000-00805F9B34FB, id: 10795158144), characteristics:
Optional("[Characteristic(uuid: 50000003-0000-1000-8000-00805F9B34FB, id: 10749750912), Characteristic(uuid: 50000002-0000-1000-8000-00805F9B34FB, id: 10749749184)]"),
error: nil)
[RxBLEKit|DEBG|08:43:23.852]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverDescriptors(
for: Characteristic(uuid: 50000003-0000-1000-8000-00805F9B34FB, id: 10749750912))
[RxBLEKit|DEBG|08:43:23.855]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverDescriptors(
for: Characteristic(uuid: 50000002-0000-1000-8000-00805F9B34FB, id: 10749749184))
[RxBLEKit|DEBG|08:43:23.857]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverCharacteristicsFor(for:
Service(uuid: FE59, id: 10795121152), characteristics:
Optional("[Characteristic(uuid: 8EC90003-F315-4F60-9FB8-838830DAEA50, id: 10749751392)]"),
error: nil)
[RxBLEKit|DEBG|08:43:23.858]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) discoverDescriptors(
for: Characteristic(uuid: 8EC90003-F315-4F60-9FB8-838830DAEA50, id: 10749751392))
[RxBLEKit|DEBG|08:43:23.859]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 50000003-0000-1000-8000-00805F9B34FB, id: 10749750912), descriptors:
Optional("[]"),
error: nil)
[RxBLEKit|DEBG|08:43:23.861]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 50000002-0000-1000-8000-00805F9B34FB, id: 10749749184), descriptors:
Optional("[Descriptor(uuid: Client Characteristic Configuration, id: 10772425360)]"),
error: nil)
[RxBLEKit|DEBG|08:43:23.865]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 8EC90003-F315-4F60-9FB8-838830DAEA50, id: 10749751392), descriptors:
Optional("[Descriptor(uuid: Client Characteristic Configuration, id: 10772519104)]"),
error: nil)
2020-07-02 09:43:23.869 [info][tid:com.facebook.react.JavaScript] discovered
2020-07-02 09:43:23.876 [info][tid:com.facebook.react.JavaScript] writing
[RxBLEKit|DEBG|08:43:23.880]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) writeValue(
data: 0101,
for: Characteristic(uuid: 50000003-0000-1000-8000-00805F9B34FB, id: 10749750912),
type: withResponse)
[RxBLEKit|DEBG|08:43:23.931]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didWriteValueFor(for:Characteristic(uuid: 50000003-0000-1000-8000-00805F9B34FB, id: 10749750912),
value: Optional(""),
error: nil)
2020-07-02 09:43:23.934 [info][tid:com.facebook.react.JavaScript] monitoring
[RxBLEKit|DEBG|08:43:23.938]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) setNotifyValue(
enabled: true,
for: Characteristic(uuid: 50000002-0000-1000-8000-00805F9B34FB, id: 10749749184))
[RxBLEKit|DEBG|08:43:23.990]: Peripheral(uuid: DCC17464-FEFC-DB10-D46A-2252F62AD9E4, name: Optional("xxx")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 50000002-0000-1000-8000-00805F9B34FB, id: 10749749184), isNotifying: true,
error: nil)
```

dariuszseweryn, 2020-07-02:
I guess you must consult your peripheral specification and figure out what you have to do. I see that monitoring starts successfully.

williamgoodhew, 2020-07-02:
I know that this is not under your questions but if you can help I would be really appreciative and if you don't want to help I totally understand.

I am trying to encrypt the key in AES128-ECB and have to maintain the big endian protocol.

So far my code to encrypt is below - but I don't think it is maintaining big endian.

I'm using the aes-js npm module.

```
var key = Buffer.from("20572F52364B3F473050415811632D2B", "hex")
    const text = '0x060x010x010x01'
    var textBytes = aesjs.utils.utf8.toBytes(text);
    console.log('textBytes: ', textBytes)

    var aesEcb = new aesjs.ModeOfOperation.ecb(key);
    console.log('aesEcb: ', aesEcb)
    var encryptedBytes = aesEcb.encrypt(textBytes);
    console.log('encryptedBytes: ', encryptedBytes)

    var encryptedHex = aesjs.utils.hex.fromBytes(encryptedBytes);
    console.log('encryptedHex: ', encryptedHex);
```

dariuszseweryn, 2020-07-02:
I have no experience with this crypto library unfortunately

williamgoodhew, 2020-07-02:
ok thanks for looking

closed 2020-07-02 by williamgoodhew