dotintent/react-native-ble-plx #708: Bug Report

stami, 2020-07-03:
Hi,

I encountered an unknown error while monitoring for a characteristic.

```
{
  "message": "Unknown error occurred. This is probably a bug! Check reason property.",
  "errorCode": 0,
  "attErrorCode": null,
  "iosErrorCode": null,
  "androidErrorCode": null,
  "reason": "k.n.c",
  "name": "BleError",
  "line": 101,
  "column": 244,
  "sourceURL": "index.android.bundle"
}
```

### Steps to Reproduce

Not reproducible. This has been reported to me just once.

### Context

* Library version: 1.1.1
* Platform: Android

Device info:
```
{
  "manufacturer": "HMD Global",
  "brand": "Nokia",
  "model": "Nokia 7 plus",
  "deviceId": "SDM660",
  "systemVersion": "10",
  "androidApiLevel": 29
}
```

The error happened while syncing data from our device. We are requesting data in ~30kB chunks and the device updates the characteristic that we are monitoring. It has been very robust for a long time now.

The error was received in `device.monitorCharacteristicForService` callback.

When encountering errors we disconnect the device, and after reconnecting the rest of the data was read successfully. So there are no means of reproducing this, unfortunately.

Do you have any idea what the `reason: "k.n.c"` may refer to? 🤞 😄

dariuszseweryn, 2020-07-03:
> Do you have any idea what the reason: "k.n.c" may refer to? 🤞 😄

By any chance you run it on an obfuscated version of the app?

stami, 2020-07-05:
Well, yeah, the log output is from a release build of the app.

dariuszseweryn, 2020-07-05:
Do you have the mapping file?

stami, 2020-07-28:
Sorry about the delay. We have a summer vacation season going on and I'll have to wait for another team member to be available for a chat about this. I'm not really familiar with native Android parts and I'm not actually even sure if the release build is _obfuscated_ or what mapping file you are referring to.

I'll get back to you later hopefully next month. This is not a blocker for us, I just reported this here as I hadn't seen anything like this before. 🙂

closed 2020-10-04 by stale[bot]