dotintent/react-native-ble-plx #709: [iOS] it seems BLE Manager cannot be restarted when refreshing the app.

azuxx, 2020-07-08:
## Expected Behaviour

Maybe BLE manager should be restarted somehow? On Android, I don't have this issue, app works properly.

## Current Behaviour

Some log issues are displayed in ios log when doing 'reset device app feature' (explained below). And app got stuck in 'looking for devices' because of scanAndConnect does not stop anymore.

### Steps to Reproduce

The following are screenshots of my app:

1) First connection (which succeeds)

[image]

2)In the app I have a 'reset' button which deletes saved device from backend, logout user, refresh the app let it doing a new scan for devices (so BLEManager is created again. See code below)

[image]

Here app seems stuck because of scanAndConnect starts again but it does not find any device even though device is On.
Same log issues are print iOS side:

```
020-07-07 18:00:38.940 [warn][tid:main][RCTView.m:853] (ADVICE) View #2197 of type RCTView has a shadow set but cannot calculate shadow efficiently. Consider setting a background color to fix this, or apply the shadow to a more specific component.
2020-07-07 18:00:40.718 [info][tid:com.facebook.react.JavaScript] called reloading of user data as per device changes happened
2020-07-07 18:00:41.256 [info][tid:main][RCTCxxBridge.mm:1070] Invalidating <RCTCxxBridge: 0x1056294f0> (parent: <RCTBridge: 0x280426140>, executor: (null))
2020-07-07 18:00:41.259 [info][tid:com.facebook.react.ShadowQueue][RNGestureHandlerManager.m:170] [GESTURE HANDLER] Tearing down gesture handler registered for views {(
    <RCTRootContentView: 0x103e797c0; reactTag: 11; frame = (0 0; 375 667); gestureRecognizers = <NSArray: 0x282145830>; layer = <CALayer: 0x282f01b60>>
)}
2020-07-07 18:00:41.261996+0200 MyApp[1056:334266] [CoreBluetooth] XPC connection invalid
2020-07-07 18:00:41.266330+0200 MyApp[1056:334687] [BoringSSL] nw_protocol_boringssl_get_output_frames(1301) [C25.1:2][0x10563d0d0] get output frames failed, state 8196
2020-07-07 18:00:41.266583+0200 MyApp[1056:334687] [BoringSSL] nw_protocol_boringssl_get_output_frames(1301) [C25.1:2][0x10563d0d0] get output frames failed, state 8196
2020-07-07 18:00:41.267368+0200 MyApp[1056:334687] TIC Read Status [25:0x0]: 1:57
2020-07-07 18:00:41.267459+0200 MyApp[1056:334687] TIC Read Status [25:0x0]: 1:57
2020-07-07 18:00:43.682565+0200 MyApp[1056:334266] [] nw_socket_handle_socket_event [C32:1] Socket SO_ERROR [61: Connection refused]
2020-07-07 18:00:43.686243+0200 MyApp[1056:334266] [] nw_connection_get_connected_socket [C32] Client called nw_connection_get_connected_socket on unconnected nw_connection
2020-07-07 18:00:43.686573+0200 MyApp[1056:334266] TCP Conn 0x281a26700 Failed : error 0:61 [61]
2020-07-07 18:00:44.633 [info][tid:com.facebook.react.JavaScript] 'has Hermes', true
2020-07-07 18:00:44.711 [info][tid:main][RCTRootView.m:294] Running application MyAppReactNative ({
    initialProps =     {
    };
    rootTag = 21;
})
2020-07-07 18:00:44.813 [info][tid:com.facebook.react.JavaScript] Running "MyAppReactNative" with {"rootTag":21,"initialProps":{}}
2020-07-07 18:00:44.920 [info][tid:com.facebook.react.JavaScript] 'Language set is', 'it'
2020-07-07 18:00:44.927 [info][tid:com.facebook.react.JavaScript] 'Initial notification was:', 'N/A'
2020-07-07 18:00:44.932 [error][tid:com.facebook.react.JavaScript] { domain: 'NSCocoaErrorDomain',
  localizedDescription: 'non è stata trovata nessuna stringa di autorizzazione “aps-environment” valida per l\'applicazione',
  code: 3000 }
2020-07-07 18:00:45.395 [info][tid:main][RNGestureHandlerManager.m:136] [GESTURE HANDLER] Initialize gesture handler for root view <RCTRootContentView: 0x103ec8950; reactTag: 21; frame = (0 0; 375 667); gestureRecognizers = <NSArray: 0x2821574e0>; layer = <CALayer: 0x282f84e80>>
2020-07-07 18:00:45.415 [info][tid:com.facebook.react.JavaScript] Fetch user data
2020-07-07 18:00:45.861 [info][tid:com.facebook.react.JavaScript] 'Got user data', { me: 
   { __typename: 'User',
     id: '30',
     confirmedAt: '2020-07-03T14:18:15Z',
     email: 'zzzzz@gmail.com',
     name: 'zzzzz',
     bridgeDevices: [] } }
2020-07-07 18:00:46.228 [info][tid:com.facebook.react.JavaScript] created BLE manager
2020-07-07 18:00:46.229 [info][tid:com.facebook.react.JavaScript] checking for existing device data...
2020-07-07 18:00:46.229 [info][tid:com.facebook.react.JavaScript] no device data found. Start device pairing procedure...
2020-07-07 18:00:46.242525+0200 MyApp[1056:334328] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x2834cb900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-07-07 18:00:46.252 [info][tid:com.facebook.react.JavaScript] 'Device Bluetooth state changed to ', 'Unknown'
2020-07-07 18:00:46.257 [info][tid:com.facebook.react.JavaScript] 'Device Bluetooth state changed to ', 'PoweredOn'
2020-07-07 18:00:46.259 [info][tid:com.facebook.react.JavaScript] do scanAndConnect
(and from here app is stuck in findind devices...)
```

So I did have some attempts to understand this issue. The most useful one is that I did a hard reset of my device:
[... 323 lines cut ...]
    "jest": "^24.9.0",
    "metro-react-native-babel-preset": "^0.58.0",
    "prettier": "^2.0.2",
    "react-native-dotenv": "^0.2.0",
    "react-test-renderer": "16.11.0",
    "typescript": "^3.8.3"
  },
  "resolutions": {
    "graphql": "^14.5.8"
  },
  "postinstall": "npx jetify",
  "jest": {
    "preset": "react-native",
    "moduleFileExtensions": [
      "ts",
      "tsx",
      "js",
      "jsx",
      "json",
      "node"
    ]
  }
```

* Formatted code sample or link to a repository: see above

dariuszseweryn, 2020-07-08:
> 2020-07-07 18:00:46.242525+0200 MyApp[1056:334328] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x2834cb900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

azuxx, 2020-07-08:
> > 2020-07-07 18:00:46.242525+0200 MyApp[1056:334328] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x2834cb900> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

@dariuszseweryn what does it mean? How can I implement this restore identifier? Sorry but I cannot find any documentation of this

dariuszseweryn, 2020-07-09:
I am surprised. What documentation can you find?

azuxx, 2020-07-09:
> I am surprised. What documentation can you find?

@dariuszseweryn I tried to use it based on this documentation but I don't know what code should I write inside in order to "deallocate" current BLE Manager instance before redoing the code which creates BLEManager again.

```javascript
if (!_bleManager.current) {
      console.log('created BLE manager');
      _bleManager.current = new BleManager(
        // {
        //   restoreStateIdentifier: 'appBleBackgroundMode',
        //   restoreStateFunction: bleRestoredState => {
        //     console.log(bleRestoredState);
        //   }
        // }
      );
    }
```

The documentation:

[image]

Do you know how can I call this
```javascript
const connectedDevice: Device = getConnectedDevice();
if (connectedDevice) {
      connectedDevice.cancelConnection();
}
RNRestart.Restart();
```
without having this issue with BLEManager?  On Android it does not happen.

dariuszseweryn, 2020-07-09:
There is API to 'deallocate/destroy' a BleManager. And you should be able to find it in the documentation. Restoring state is for just that — restoring state, and it is documented as well. If you want to persist or recreate the manager because of RN app reloads you need to ask on RN forums as this is not an issue with this library.

azuxx, 2020-07-09:
@dariuszseweryn I did in this way and it worked, thank you!

```javascript
if (appState.connectedDevice) {
        appState.connectedDevice.cancelConnection();
}
 _bleManager.current.destroy();
 _isMounted.current = false;
 delete _bleManager.current;
 RNRestart.Restart();
```

closed 2020-07-09 by azuxx