dotintent/react-native-ble-plx #712: Question

CarbonC, 2020-07-10:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Could we please pin somewhere on this repository issues/ the readme the known issues with ble, even not specific to this lib? This one for example concerns a lot of peoples imo but is hard to find, especially if you only test with a samsung: https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509/75 When testing the library i was stucked because of that and thought it was the libraries' fault...

hooksoft, 2020-07-14:
Also, don't use react-native 0.63.0...

nriccar, 2020-07-27:
Also, is not working for IOS. Just for Android

CarbonC, 2020-07-27:
@nriccar Im using it on production on IOS. If you have any troubles just fill another issue please, the point of this question is not to list all difficulties you can have on installation but to ask for a list of KNOWN bugs to be accessible.

nriccar, 2020-07-27:
@CarbonC Its not actually a "difficulty", its an error. Polidea confirmed themselves. You do have a production IOS working code that is retrieving services and characteristics from an iPhone?  

This is strange, since Polidea confirmed me that characteristics for IOS were not working. If you could share your package.json it would be really useful

CarbonC, 2020-07-27:
> @CarbonC Its not actually a "difficulty", its an error. Polidea confirmed themselves. You do have a production IOS working code that is retrieving services and characteristics from an iPhone?
>
> This [...]

@nriccar Im using react native ble plx 2.0.0... But i dont think im actually retrieving characteristics or service from the remote device since I connect only to a hardware that is under my control and i know what is exposed. If characteristics scan is a known bug on IOS it clearly deserves to be on a list accessible without pm to Polidea...

SunstroUS, 2020-08-06:
@nriccar Where did they confirm this at? I was able to scan, discover and read from a characteristic on iOS with 
```
...
"react-native": "~0.62.2",
"react-native-ble-plx": "^2.0.1"
...
```

nriccar, 2020-08-06:
They confirmed it via email and on a private call. I´ll check again, maybe its from my device, although the main engineer behind the libraries confirmed that he was aware of this issues..

closed 2020-10-17 by stale[bot]

hooksoft, 2020-12-02:
i'm not getting updates for my monitorCharacteristicForService callback on iOS - should I try an older version of thelibrary? I'm on the most recent version.