dotintent/react-native-ble-plx #713: not find compatible versions for pod "MultiplatformBleAdapter"

hamzadahmani, 2020-07-15:
[!] CocoaPods could not find compatible versions for pod "MultiplatformBleAdapter":
  In snapshot (Podfile.lock):
    MultiplatformBleAdapter (= 0.1.6)

  In Podfile:
    react-native-ble-plx (from `../node_modules/react-native-ble-plx`) was resolved to 2.0.1, which depends on
      MultiplatformBleAdapter (= 0.1.6)

Specs satisfying the `MultiplatformBleAdapter (= 0.1.6), MultiplatformBleAdapter (= 0.1.6)` dependency were found, but they required a higher minimum deployment target.

dariuszseweryn, 2020-07-15:
I do not get it. The last time podspec for `MultiplatformBleAdapter` was changed in commit `f0d06265176433405e3d1a9c855ec5b7584e2289` of `MultiplatformBleAdapter` which was made on 9th of December 2019 which was before `react-native-ble-plx@2.0.0`.

Could you elaborate what was working and is not working anymore?

hamzadahmani, 2020-07-15:
im upgrade from 1.1.1 to 2.1.1 then i pod install and i get this issue
[!] CocoaPods could not find compatible versions for pod "MultiplatformBleAdapter":
  In Podfile:
    react-native-ble-plx (from `../node_modules/react-native-ble-plx`) was resolved to 2.0.1, which depends on
      MultiplatformBleAdapter (= 0.1.6)

None of your spec sources contain a spec satisfying the dependency: `MultiplatformBleAdapter (= 0.1.6)`.

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

Note: as of CocoaPods 1.0, `pod repo update` does not happen on `pod install` by default.

dariuszseweryn, 2020-07-15:
Correct me if I am wrong but it seems that the pods even tells you all steps you need to perform to make it work?

hamzadahmani, 2020-07-15:
yes i try it but not working 
i downgrade to @2.0.0 its  working but the problem only on version 2.0.1 on ios

dariuszseweryn, 2020-07-15:
What have you tried? `pod repo update`?

hamzadahmani, 2020-07-15:
pod install --repo-update

dariuszseweryn, 2020-07-15:
Judging from [cocoapods](http://cocoapods.org/pods/MultiplatformBleAdapter) [podspec file](https://github.com/CocoaPods/Specs/blob/master/Specs/2/a/e/MultiplatformBleAdapter/0.1.6/MultiplatformBleAdapter.podspec.json) everything seems to correctly point to version `0.1.6`. It also seems to work on my end.

Do you have any ideas what may be wrong?

Ninjinka, 2020-07-27:
why not simply  pod update MultiplatformBleAdapter ?

hamzadahmani, 2020-07-27:
Its fine thé problème

closed 2020-07-27 by hamzadahmani

CeruleanW, 2021-11-16:
If you use an m1 apple chipset MacBook like me, use `arch -x86_64 pod repo update` instead of `pod repo update`

ngocvu576, 2023-09-30:
You should run `pod update MultiplatformBleAdapter` to apply changes you've made.

Niyamath17, 2024-03-05:
> You should run `pod update MultiplatformBleAdapter` to apply changes you've made.

Its working for me, thank you!