dotintent/react-native-ble-plx #716: error: module map file '/.../MultiplatformBleAdapter.modulemap' not found

roberto-buratti, 2020-07-17:
I'm having a big trouble with your plugin setup.

I was using 1.1.0 till yesterday and all was working fine.
After having upgrade xcode to 11.6 I wasn't more able to compile my project.
So, I tried to upgrade to plugin to 2.0.1 but now I get this error:

**error: module map file '/.../Library/Developer/Xcode/DerivedData/xxx-eyicbfvltgytoaajmpvheqagxbhy/Build/Products/Release-iphoneos/MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found**

Please help me to fix this issue

dariuszseweryn, 2020-07-23:
Have you gone through migration procedure and run `pod install`?

ankitgusai, 2020-08-07:
I am facing the same issue. 
I have followed the procedure mentioned here: https://github.com/Polidea/react-native-ble-plx#ios-example-setup
I am running xcode 11.5
react-native-ble-plx version 2.0.1

The error is : 
 **module map file '/.../Library/Developer/Xcode/DerivedData/AwesomeProject-cmcxrnertsehnadwgrwewpctmhyw/Build/Products/Debug-iphonesimulator/MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found

failed to emit precompiled header '/.../Library/Developer/Xcode/DerivedData/AwesomeProject-cmcxrnertsehnadwgrwewpctmhyw/Build/Intermediates.noindex/PrecompiledHeaders/AwesomeProject-Bridging-Header-swift_2W1ENYSBLE5D-clang_1BGZMS7NMOI9O.pch' for bridging header '/.../ReactNativeProjects/TestForDummies/AwesomeProject/ios/AwesomeProject-Bridging-Header.h'**

ankitgusai, 2020-08-07:
After a bit of digging, I managed to figure out that this issue is only happening when I open project.xcodeproj but not when I open project.xcworkspace in xcode. 

I am a novice RN developer and don't have any experience with iOS development. I suppose when you said '_Open Xcode workspace_' in the 3rd point in the iOS setup you meant open project.xcworkspace. 

My apologies.

jahead, 2020-09-30:
We are seeing this issue with `yarn run ios`  on 0.62.x but not with building it from xcode by opening the workspace.

I can fix this by adding a .swift file, but doesn't seem to be in the spirit of autolinking. do we still add the bridging file?

jahead, 2020-09-30:
Okay, with trying to debug this, I was able to get this working. 

Deleting the pods folder didn't seem to work. 
I had to delete the node_modules/Pods folder/pod cache and `DerivedData` folder. 

it was still broken, but after rebooting it worked.
I have no idea what happened, but `yarn run ios` was definitely caching something, and doing a simple pods/node_modules reinstalling didn't work.

phithu, 2020-10-30:
I also faced this issue when I run on simulator. However it's working fine on real device.
My environment:
- iOS: 14.1
- Xcode: 12.1
- React Native: 0.63.3

closed 2021-01-11 by stale[bot]

andreas-arkulpa, 2021-04-20:
i am having the same issue. what was the steps you have done in order to get it working?

twwspes, 2021-04-21:
My app had once got a similar error after upgrading the bare workflow expo app's expo sdk version from 40 to 41.  After manually adjusting many node_modules and pod install and pod updates, I got stuck to the same issue:

error: module map file '/.../Library/Developer/Xcode/DerivedData/xxx-eyicbfvltgytoaajmpvheqagxbhy/Build/Products/Release-iphoneos/MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found

But it was accompanied with another issue, which made me think that the underlying issue was not the above issue:

error: failed to emit precompiled header '/var/folders/kd/4gh0_kxx3jx4thjb_sssmmcw0000gn/T/AppName-Bridging-Header-97bd5f.pch' for bridging header '/Users/ringo/Downloads/AppName-master/AppName/Resources/AppName-Bridging-Header.h'

It only happened when a release version was built.  When I built it in debug, everything is just fine.

Then I search for the reason for "error: failed to emit precompiled header" and got a bunch of solutions for this issue here:

https://stackoverflow.com/questions/45779591/failed-to-emit-precompiled-header-for-bridging-header/52721354

Here I found my solution is that I added x86_64 in Xcode's "Project -> Build Settings -> User-Defined -> VALID_ARCHS", it turns out to be that the system did not recompile the app in simulator, which runs intel's x86_64 arch.

If VALID_ARCHS cannot be found, just add it by clicking the + on the tabs above.

Still don't know will it affect build for real devices

twwspes, 2021-04-22:
A New update for my post yesterday.   the use of VALID_ARCHS is deprecated since Xcode 12 so we'd better not touching it.

Instead, if you are having the problem when building for simulator and you are not using apple M1 laptop, you may add arm64 to "Excluded Architecture - Any iOS Simulator SDK" under "build settings" on both your target app and pod.

But my case is about that after I upgrade the pods and apps with ios deployment target to 11.0 under "Info", I forget to adjust it under "Build Settings - Deployment - iOS Deployment Target".

DBoroujerdi, 2021-06-16:
I'm running an M1 mac and fixed this issue by starting the simulator using Rosetta. Try adding `arch -x86_64` before the command you use to start your app in a simulator.

e.g
```
arch -x86_64 react-native run-ios --simulator="iPhone 12"
```

It makes the build dog slow but at least it works. This issue has been bugging me for weeks.

Does anyone know why this works?

diegobonagurio, 2022-07-21:
> Depois de pesquisar um pouco, consegui descobrir que esse problema só está acontecendo quando abro o project.xcodeproj, mas não quando abro o project.xcworkspace no xcode.
> 
> Eu sou um desenvolvedor RN iniciante e não tenho nenhuma experiência com desenvolvimento iOS. Suponho que quando você disse ' _Open Xcode workspace_ ' no 3º ponto na configuração do iOS você quis dizer open project.xcworkspace.
> 
> Me desculpe.

Its work for me. Tks

KatyeStevens, 2022-10-31:
> Okay, with trying to debug this, I was able to get this working.
>
> Deleting the pods folder didn't seem to work. I had to delete the node_modules/Pods folder/pod cache and `DerivedData` folder.
>
> it [...]

Thank you so much. Seriously. You are my freaking hero.