dotintent/react-native-ble-plx #717: Question

Madhu02, 2020-07-21:
## Question

How to read BLE Peripheral Characteristics in react native using react-native-ble-plx

I'm developing an application which scans for BLE devices(Peripherals), then connect to them, read their services and characteristics. For now, I was able to connect to an iPhone which acting as a Peripheral device with certain characteristics and also was able to connect to Fitbit Versa Watch to read the heart sensor data. Although I was able to discover their services, I wasn't able to extract more information from the services like ServiceUUId and their characteristics.

Below is my code is written in react native.

    scanAndConnect() {
    console.log("Scanning Started");
    this.manager.startDeviceScan(null, null, (error, device) => {
    if (error) {
      // Handle error (scanning will be stopped automatically)
     console.log("Error in scanning devices:", error);
     return
     }
       // Check if it is a device you are looking for based on advertisement data
       // or other criteria.
      console.log("Detected Device Details:", device.id, device.name);
       // ||device.localName === 'BLEPeripheralApp') 
       if (device.name === 'Versa Lite'){ //
          // Stop scanning as it's not necessary if you are scanning for one device.
          console.log("Device Found, Stopping the Scan.");
          console.log("Connecting to:",device.name)
          this.manager.stopDeviceScan();
          device.connect()
          .then((device) => {
           // this.info("Discovering services and characteristics")
           console.log("Connected...Discovering services and characteristics");
         return device.discoverAllServicesAndCharacteristics()
       })
      .then((device) => {
        console.log('Services and characteristics discovered');
        //return this.testChar(device)
        const services = device.services()
        console.log(services);
        return device.readCharacteristicForService(services)
        // device.readCharacteristicForService("abbaff00-e56a-484c-b832-8b17cf6cbfe8")
        // this.info("Setting notifications")
        //return this.setupNotifications(device)
      })
      .then(() => {
        const characteristicsData =  device.readCharacteristicForService();
        console.log(characteristicsData);
        //this.info("Listening...")
      }, (error) => {
        console.warn(error.message);
        // this.error(error.message)
        })
       }
     });
    }

How do I extract the serviceUUId from the Service method and also read the characteristics of that service? For my iPhone peripheral, I have two mutable characteristics that I should be able to read and write. How do I read them from service to characteristics to the real value?

Any help/suggestions are much appreciated.

Thanks.

dariuszseweryn, 2020-07-23:
Do you face any issues?

Madhu02, 2020-07-27:
@dariuszseweryn Thank you for responding. I was able to get the services and characteristics now. Have modified the above code and seems working fine. But i have a question how do i read the data of characteristics of each service. I have Fitbit Versa Lite Watch trying to connect to it and read the heart rate. I was able to fetch the services and its characteristics, but not getting how do i read the characteristics value. We have read, write, writeWithOutResponse, writeWithResponse i'm confused with this. if i want to read the heart rate of a watch how exactly do i need to proceed once the services and characteristics has been discovered. Kindly Help. Thank you.

lnvtphu, 2020-08-12:
I've got the same issue. I've read the GATT Services (https://www.bluetooth.com/specifications/gatt/services/). But I don't know how to get the specified service value like Heart Rate, Running Speed and Cadence, etc.

Madhu02, 2020-08-12:
@lnvtphu It all depends on the device which you are trying read the characteristics, some of the devices like (Samsung Gear, Fibit Versa Lite)doesn't expose their heart rate services and characteristics. I was able to read the characteristics, but couldn't find the one related to Heart Rate as they aren't exposing their services. But Make you have 0*180D in any on the services UUID.

Below code should help you getting the characteristics for the services.

async readData(device) {
        const services = await device.services();
        console.log("Services:",services);
        const characteristics = await services[1].characteristics();
        console.log("Characteristics:",characteristics);
        characteristics[0].monitor((err, update) => {
          if (err) {
            console.log(`characteristic error: ${err}`);
            console.log(JSON.stringify(err));
          } else {
            console.log("Is Characteristics Readable:",update.isReadable);
            console.log("Heart Rate Data:",base64.decode(update.value));
            // const readCharacteristic = await device.readCharacteristicForService(userDataServiceUUID, heightCharacteristicUUID); // assuming the device is already connected
            // var data = new Uint16Array(base64.decode(update.value));

            const heartRateData = Buffer.from(update.value, 'base64').readUInt16LE(0);
            console.log("Heart Beats:",heartRateData);
          }
        });
      }

lnvtphu, 2020-08-14:
@Madhu02 many thanks. At the moment, I try to write the Native Module to integrate the Band SDK provide by customer.

FazilMuhammed, 2020-10-07:
@Madhu02 Hello Brother..
if you have idea about to connect bluetooth printer and print the slip.
i have issue with bluetooth printer i cannot write data.please help me to get print slip

closed 2020-12-18 by stale[bot]

keerthi552, 2022-01-03:
iam retetrieving characteristics of a service in that characteristics object iam getting value property as null("value":null) 
how can i get the value

Tadejo9, 2022-01-03:
@keerthi552 
Have you called discoverServicesAndCharacteristics?

keerthi552, 2022-01-03:
yes i have called discoverServicesAndCharacteristics   i got the services and characteristics,but with the characteristics value as null

Tadejo9, 2022-01-03:
Which method did you use to read data from the characteristic?

keerthi552, 2022-01-03:
BleManager.characteristicsForDevice(device id,service uuid)

the result is
 [{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 12}, "deviceID": device id, "id": 4, "isIndicatable": true, "isNotifiable": true, "isNotifying": false, "isReadable": false, "isWritableWithResponse": true, "isWritableWithoutResponse": false, "serviceID": 3, "serviceUUID": serviceId, "uuid": uuid, "value": null}]

Tadejo9, 2022-01-03:
https://dotintent.github.io/react-native-ble-plx/

```

characteristicsForDevice(deviceIdentifier, serviceUUID)
List of discovered Characteristics for given Device and Service.

Promise<Array<Characteristic>>: Promise which emits array of Characteristic objects which are discovered for a Device in specified Service .
```

```
readCharacteristicForService(serviceUUID, characteristicUUID, transactionId)
Promise<Characteristic>: Promise which emits first Characteristic object matching specified UUID paths. Latest value of Characteristic will be stored inside returned object.
```

keerthi552, 2022-01-04:
readCharacteristicForService is not working actually, instead i used characteristicsForService this returned object same as the one that returned by the method characteristicsForDevice,

if we want a device information like minPressure,maxPressure,ramp......, then, is these will we available in characteristics by id or it will be with a property name as these  minPressure,maxPressure,ramp......

and if for every property(minPressure,maxPressure,ramp......)  if it is obtained by characteristic id then iam not getting array of characteristics iam getting only one characteristic object

Madhu02, 2022-01-04:
Hi @keerthi552 
Could you please elaborate what are you trying to achieve. I have working code of reading and writing data to and from characteristics to the device using this library.

keerthi552, 2022-01-05:
> Hi @keerthi552 Could you please elaborate what are you trying to achieve. I have working code of reading and writing data to and from characteristics to the device using this library. Which device are you trying to connect ?

1)i want data like min Pressure,max Pressure,ramp,ramp time from a device,by refering some codes i understood that these data (Pressure,max Pressure,ramp,ramp) cannot be obtained directly instead we get the service uuid's,characeristic uuid's based on these uuid's how can we know that the characteristic uuid belongs to some pressure's  or something else
so,let me know if there is any way to convert characteristic uuid to meaning full characteristic string

2)even though i got an characteristic object iam getting the value property in characteristic object as null

Madhu02, 2022-01-05:
> 1)i want data like min Pressure,max Pressure,ramp,ramp time from a device,by refering some codes i understood that these data (Pressure,max Pressure,ramp,ramp) cannot be obtained directly instead we [...]

First of all which device you are trying to connect?
What is the device BLE Profile ?
Does your BLE device has its own ble identifiers or does it follow Bluetooth standard specifications as per this link https://www.bluetooth.com/specifications/assigned-numbers/
Without knowing what your BLE peripheral device details and their identifiers, you cannot just connect and read those values. 
Does your BLE device has min Pressure,max Pressure,ramp,ramp time attributes in their profile ?
For more information on BLE, go through the below links and see if your device brand or manufacture has certain assigned number for the attributes or values you are looking for.
https://www.bluetooth.com/specifications/assigned-numbers/
https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/

taj567, 2022-01-05:
Hi Guys, am using react-native-ble-plx && react-native-Bluetooth-serial using this packages am not able to retrieve the data from a paired Bluetooth device called PHILIPS CPAP able to retrieve characteristics services & descriptors. not able to get the current data from it. -Thanks in Advance

------data receiving ------
	true status if the device connected it returns true

LOG 						 *******services*************

LOG  	found services:  [
{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], 
"_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 9}, "deviceID": "00:1F:FF:49:C8:6D", "id": 1, "isPrimary": true, 
"uuid": "00001801-0000-1000-8000-00805f9b34fb"}, {"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], 
"_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 9}, 
"deviceID": "00:1F:FF:49:C8:6D", "id": 3, "isPrimary": true, "uuid": "4553867f-f809-49f4-aefc-e190a1f459f3"}]
 
LOG  					*********services[0]._manager._eventEmitter************

LOG  {"_subscriber":
	{"_currentSubscription": null, 
	"_subscriptionsForType": {"ScanEvent": [Array], 
	"StateChangeEvent": [Array], 
	"appStateDidChange": [Array], 
	"collectBugExtraData": [Array], 
	"collectRedBoxExtraData": [Array],
	"didUpdateDimensions": [Array], 
	"hardwareBackPress": [Array],
	"keyboardDidHide": [Array], 
	"keyboardDidShow": [Array], 
	"keyboardWillHide": [Array], 
	"keyboardWillShow": [Array], 
	"toggleElementInspector": [Array], 
	"websocketClosed": [Array], 
	"websocketFailed": [Array], 
	"websocketMessage": [Array], 
	"websocketOpen": [Array]}}}

					*********services._manager._eventEmitter._subscriber************

LOG  {"_currentSubscription": null, "_subscriptionsForType": {"ScanEvent": [[EmitterSubscription], undefined],
	 "StateChangeEvent": [undefined, undefined], 
	 "appStateDidChange": [[EmitterSubscription]],
	 "collectBugExtraData": [[EmitterSubscription]], 
	 "collectRedBoxExtraData": [[EmitterSubscription]],
	 "didUpdateDimensions": [[EmitterSubscription]], 
	 "hardwareBackPress": [[EmitterSubscription]],
	 "keyboardDidHide": [undefined, [EmitterSubscription], undefined], 
	"keyboardDidShow": [undefined, [EmitterSubscription], undefined],
	 "keyboardWillHide": [undefined, [EmitterSubscription], undefined], 
	"keyboardWillShow": [undefined, [EmitterSubscription], undefined],
	"toggleElementInspector": [[EmitterSubscription]],
	"websocketClosed": [undefined, [EmitterSubscription]],
	"websocketFailed": [undefined, [EmitterSubscription]], 
	"websocketMessage": [undefined, [EmitterSubscription]], 
	"websocketOpen": [undefined, [EmitterSubscription]]}}

LOG			 	*********services._manager._eventEmitter._subscriber._subscriptionsForType************

LOG  {"ScanEvent": [undefined, undefined], "StateChangeEvent": [undefined, undefined], 
	"appStateDidChange": [{"context": undefined, "emitter": [AppState], "eventType": "appStateDidChange", "key": 0, "listener": [Function anonymous], "subscriber": [EventSubscriptionVendor]}], 
	"collectBugExtraData": [{"context": null, "emitter": [RCTDeviceEventEmitter], "eventType": "collectBugExtraData", "key": 0, "listener": [Function collectExtraData], "subscriber": [EventSubscriptionVendor]}],
	"collectRedBoxExtraData": [{"context": null, "emitter": [RCTDeviceEventEmitter], "eventType": "collectRedBoxExtraData", "key": 0, "listener": [Function collectExtraData], "subscriber": [EventSubscriptionVendor]}], 
	"didUpdateDimensions": [{"context": undefined, "emitter": [RCTDeviceEventEmitter], "eventType": "didUpdateDimensions", "key": 0, "listener": [Function anonymous], "subscriber": [EventSubscriptionVendor]}], 
	"hardwareBackPress": [{"context": undefined, "emitter": [RCTDeviceEventEmitter], "eventType": "hardwareBackPress", "key": 0, "listener": [Function anonymous], "subscriber": [EventSubscriptionVendor]}], 
	"keyboardDidHide": [undefined, undefined, undefined, {"context": undefined, "emitter": [NativeEventEmitter], "eventType": "keyboardDidHide", "key": 3, "listener": [Function bound scrollResponderKeyboardDidHide], "subscriber": [EventSubscriptionVendor]}], 
	"keyboardDidShow": [undefined, undefined, undefined, {"context": undefined, "emitter": [NativeEventEmitter], "eventType": "keyboardDidShow", "key": 3, "listener": [Function bound scrollResponderKeyboardDidShow], "subscriber": [EventSubscriptionVendor]}], 
	"keyboardWillHide": [undefined, undefined, undefined, {"context": undefined, "emitter": [NativeEventEmitter], "eventType": "keyboardWillHide", "key": 3, "listener": [Function bound scrollResponderKeyboardWillHide], "subscriber": [EventSubscriptionVendor]}], 
	"keyboardWillShow": [undefined, undefined, undefined, {"context": undefined, "emitter": [NativeEventEmitter], "eventType": "keyboardWillShow", "key": 3, "listener": [Function bound scrollResponderKeyboardWillShow], "subscriber": [EventSubscriptionVendor]}], 
	"toggleElementInspector": [{"context": undefined, "emitter": [RCTDeviceEventEmitter], "eventType": "toggleElementInspector", "key": 0, "listener": [Function anonymous], "subscriber": [EventSubscriptionVendor]}], 
	"websocketClosed": [undefined, {"context": undefined, "emitter": [NativeEventEmitter], "eventType": "websocketClosed", "key": 1, "listener": [Function anonymous], "subscriber": [EventSubscriptionVendor]}], "websocketFailed": [undefined, {"context": undefined, "emitter": [Na [... cut ...]
LOG  {"_subscriber": {"_currentSubscription": null, "_subscriptionsForType": {"ScanEvent": [Array], "StateChangeEvent": [Array], 
	"appStateDidChange": [Array], 
	"collectBugExtraData": [Array], 
	"collectRedBoxExtraData": [Array], 
	"didUpdateDimensions": [Array], 
	"hardwareBackPress": [Array], 
	"keyboardDidHide": [Array], 
	"keyboardDidShow": [Array], 
	"keyboardWillHide": [Array], 
	"keyboardWillShow": [Array], 
	"toggleElementInspector": [Array], 
	"websocketClosed": [Array], 
	"websocketFailed": [Array], 
	"websocketMessage": [Array], 
	"websocketOpen": [Array]}}
     }

keerthi552, 2022-01-07:
> First of all which device you are trying to connect? What is the device BLE Profile ? Does your BLE device has its own ble identifiers or does it follow Bluetooth standard specifications as per this [...]

iam using PHILIPS CPAP device,i got 2 services,is there any way to decode the ble service uuid's to know which service they are providing

Madhu02, 2022-01-07:
You will be able to get the service and their related UUID's in the service object. But certain manufacturers won't advertise their private data considering it as a secure. I suggest you first check the device documentation.

karthick-iorbit, 2022-06-01:
@Madhu02 
Hi, I have try to integrate smartwatch(bleDevice) with react native; 
i got protocol document of smartwatch; 
now i need to get data from that smartwatch (parameters like spo2,bpm--need these data);
can u plz guide me to get data.