dotintent/react-native-ble-plx #720: onDeviceDisconnected returns null error when connection lost unexpectedly (Android)

DavidEdwardMead, 2020-07-22:
## Expected Behavior

onDeviceDisconnected event returns non-null error when connection lost unexpectedly, i.e. there is no bleManager.cancelDeviceConnection() call.

## Current Behavior

onDeviceDisconnected event returns null error whenever a connection is lost, whether or not cancelDeviceConnection() is called.

### Steps to Reproduce

1. Connect to remote device via bluetooth.
2. Call onDeviceDisconnected(deviceIdentifier, listener)
3. Cause the connection to be lost, e.g. power down remote device.
4. Log error returned from listener callback.

### Context

* Library version: 2.0.1
* Platform: Android.

### Comments

Not urgent for me, as we can work round this by setting a flag when we call cancelDeviceConnection().

dariuszseweryn, 2020-07-23:
Could you add native logs?

pmgmdm, 2020-07-23:
> Could you add native logs?

Connecting...
2020-07-23 14:04:31.931 22615-22729/com... D/BluetoothGatt: connect() - device: ..., auto: false
2020-07-23 14:04:31.931 22615-22729/com... D/BluetoothGatt: registerApp()
2020-07-23 14:04:31.931 22615-22729/com... D/BluetoothGatt: registerApp() - UUID=...
2020-07-23 14:04:31.937 22615-22651/com... D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2020-07-23 14:04:32.195 22615-22651/com... D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=...
2020-07-23 14:04:32.258 22615-22729/com... D/BluetoothGatt: discoverServices() - device: ...
2020-07-23 14:04:32.530 22615-22651/com... D/BluetoothGatt: onConnectionUpdated() - Device=... interval=6 latency=0 timeout=500 status=0
2020-07-23 14:04:33.084 22615-22651/com... D/BluetoothGatt: onSearchComplete() = Device=... Status=0
2020-07-23 14:04:33.134 22615-22651/com... D/BluetoothGatt: onConnectionUpdated() - Device=... interval=39 latency=0 timeout=500 status=0
2020-07-23 14:04:33.359 22615-22687/com... D/BluetoothGatt: setCharacteristicNotification() - uuid: ... enable: true
2020-07-23 14:04:33.378 22615-22687/com... D/BluetoothGatt: setCharacteristicNotification() - uuid: ... enable: true
2020-07-23 14:04:37.353 22615-22642/com... D/BluetoothGatt: onConnectionUpdated() - Device=... interval=78 latency=2 timeout=400 status=0

After remote device powered off...
2020-07-23 14:05:00.144 22615-22642/com... D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=6 device=...
2020-07-23 14:05:00.172 22615-22729/com... D/BluetoothManager: getConnectionState()
2020-07-23 14:05:00.172 22615-22729/com... D/BluetoothManager: getConnectedDevices
2020-07-23 14:05:00.178 22615-22642/com... D/BluetoothGatt: setCharacteristicNotification() - uuid: ... enable: false
2020-07-23 14:05:00.178 22615-22729/com... D/BluetoothGatt: close()
2020-07-23 14:05:00.178 22615-22729/com... D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-07-23 14:05:00.187 22615-22642/com... D/BluetoothGatt: setCharacteristicNotification() - uuid: ... enable: false

Thanks.

suark, 2020-09-09:
Experiencing the same issue I believe. 
Reboot my bluetooth device, which used to give a disconnection error on android, but not isn't giving one. 
I also have to work around by using a flag.

drchorney, 2020-09-29:
Yeah me too, I'm getting no errors descriptions anymore; the Error is always null. I'm forcing all sorts of disconnections from the connected device (not the react-native client). No errors. Older versions I'd get an error.

AmirHMousavi, 2020-10-05:
Also, the returned `Device` object is not complete. I just checked two that I needed, `name` & `loaclName` are missing (not `null` but the attribute is missing).

PeterF-Gluxkind, 2021-11-22:
Can confirm this issue is also present in iOS. Force reboot of my ESP32 also return 'null'.

react-native-ble-plx: 2.0.2
Xcode: 15
React: 17.0.2
RN: 0.66.0

```javascript
const ConnectionStateSubscription = manager.onDeviceDisconnected(deviceId, 
                (error, device) => {
                    console.log(error) // Bug: returns null even when ESP32 is force rebooted
                    if (error !== null) 
                        Alert.alert('Alert', 'BLE connection lost.')
                        navigation.navigate('Connect')
                    }
            })
```

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.