dotintent/react-native-ble-plx #721: How to connect to multiple devices simultaneously ?

jithinlal, 2020-07-23:
I was trying out the module for a project I was working on. Could you guide me on how we can connect to multiple devices? In particular receive updates from 3 devices and process them in the react-native application.
Thanks in advance!

dariuszseweryn, 2020-07-23:
Hello
It will work the same as for a single device:
1. Get the three devices you want to connect
2. Stop scanning (some Android devices do not handle scanning and connections well)
3. Connect first device
4. Connect second device
5. Connect third device

jithinlal, 2020-07-23:
I meant can we connect one and keep it connected while we connect the other. So we can connect to them interchangeably?

dariuszseweryn, 2020-07-23:
Yes

closed 2020-07-25 by jithinlal