dotintent/react-native-ble-plx #723: java.lang.IllegalArgumentException · null is not a valid Bluetooth address

amitind49, 2020-07-27:
> BluetoothDevice.java:1359android.bluetooth.BluetoothDevice.<init>	
> BluetoothAdapter.java:1076android.bluetooth.BluetoothAdapter.getRemoteDevice	
> RxBleAdapterWrapper.java:29com.polidea.rxandroidble.internal.util.RxBleAdapterWrapper.getRemoteDevice	
> DeviceModule.java:41com.polidea.rxandroidble.internal.DeviceModule.provideBluetoothDevice	
> DeviceModule_ProvideBluetoothDeviceFactory.java:24com.polidea.rxandroidble.internal.DeviceModule_ProvideBluetoothDeviceFactory.get	
> DeviceModule_ProvideBluetoothDeviceFactory.java:10com.polidea.rxandroidble.internal.DeviceModule_ProvideBluetoothDeviceFactory.get	
> RxBleDeviceImpl_Factory.java:31com.polidea.rxandroidble.internal.RxBleDeviceImpl_Factory.get	
> RxBleDeviceImpl_Factory.java:11com.polidea.rxandroidble.internal.RxBleDeviceImpl_Factory.get	
> DoubleCheck.java:47bleshadow.dagger.internal.DoubleCheck.get	
> DaggerClientComponent.java:480com.polidea.rxandroidble.DaggerClientComponent$DeviceComponentImpl.provideDevice	
> RxBleDeviceProvider.java:43com.polidea.rxandroidble.internal.RxBleDeviceProvider.getBleDevice	
> RxBleClientImpl.java:94com.polidea.rxandroidble.RxBleClientImpl.getBleDevice	
> BleModule.java:454com.polidea.multiplatformbleadapter.BleModule.cancelDeviceConnection	
> BleClientManager.java:355com.polidea.reactnativeble.BleClientManager.cancelDeviceConnection	
> Method.java:-2java.lang.reflect.Method.invoke	
> JavaMethodWrapper.java:371com.facebook.react.bridge.JavaMethodWrapper.invoke	
> JavaModuleWrapper.java:150com.facebook.react.bridge.JavaModuleWrapper.invoke	
> NativeRunnable.java:-2com.facebook.react.bridge.queue.NativeRunnable.run	
> Handler.java:883android.os.Handler.handleCallback	
> Handler.java:100android.os.Handler.dispatchMessage	
> MessageQueueThreadHandler.java:26com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage	
> Looper.java:237android.os.Looper.loop	
> MessageQueueThreadImpl.java:225com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run	
> Thread.java:919java.lang.Thread.run

gidagel, 2020-08-04:
Facing the same issue. Anyone who knows what it its?

MattJakeman, 2020-09-17:
I've just had this same issue as well and no idea as to what is causing it.

closed 2020-11-29 by stale[bot]

itseramin, 2022-05-12:
Same issue but only on android and on release mode...

ShongSu, 2023-06-01:
hello @itseramin, did you resolve it? I am facing the same problem.