dotintent/react-native-ble-plx #733: Bug Report

velthune, 2020-08-26:
## Prerequisites

- [ V] I am running the latest version
- [ V] I checked the documentation and found no answer
- [ V] I checked to make sure that this issue has not already been filed

## Expected Behavior

Receiving  notification on sending command to device

## Current Behavior

DO NOT receive any kind of notification on sending command to device

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. connecting device using:
 `bleManager.connectToDevice(device.id)`
2. once connected (promise resolve) I discover device info:
```
discoverAllInfo(device)
        .then(getServices)
        .then(findNordicService)
        .then((service?: Service) => service && getCharacteristics(service))
        .then((characteristic?: Characteristic[]) => characteristic && getRxTx(characteristic))
```
This method simply returns an object with **RX** and **TX** characteristics on promise resolve.

3. Once device info have been discovered, I'm start monitoring RX channel using:
`rx.monitor`

4. Send a command to the device using this util method:

```
export const sendCommand = (characteristic?: Characteristic, value?: string, withoutResponse?: boolean) => {
    const text = (value || "") + '\r\n';
    const command = base64.encode(text);
    return withoutResponse ? characteristic?.writeWithoutResponse(command) : characteristic?.writeWithResponse(command)
}
```
### Context

* Library version: "react-native-ble-plx": "^2.0.1"
* Platform: iOS.
* Platform logs (logcat/XCode):

```
[RxBLEKit|DEBG|08:48:43.515]: CentralManager(10772837168) cancelPeripheralConnection(
[RxBLEKit|DEBG|08:48:43.517]: CentralManager(10772837168) didDisconnect(from: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")),
[RxBLEKit|DEBG|08:48:43.561]: CentralManager(10772885760) didUpdateState(state: poweredOn)
[RxBLEKit|DEBG|08:48:43.667]: CentralManager(10772885760) retrievePeripherals(
[RxBLEKit|DEBG|08:48:43.668]: CentralManager(10772885760) connect(
[RxBLEKit|DEBG|08:48:43.730]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverServices(
[RxBLEKit|DEBG|08:48:43.731]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverServices(services
[RxBLEKit|DEBG|08:48:43.731]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverCharacteristics(
[RxBLEKit|DEBG|08:48:43.732]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|08:48:43.732]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
[RxBLEKit|DEBG|08:48:43.733]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
[RxBLEKit|DEBG|08:48:43.734]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
[RxBLEKit|DEBG|08:48:43.734]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
[RxBLEKit|DEBG|08:48:43.760]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) setNotifyValue(
[RxBLEKit|DEBG|08:48:43.763]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) writeValue(
[RxBLEKit|DEBG|08:48:43.788]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) writeValue(
[RxBLEKit|DEBG|08:48:43.800]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didUpdateNotificationStateFor(
[RxBLEKit|DEBG|08:48:43.801]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) peripheralIsReady(toSendWriteWithoutResponse:true
[RxBLEKit|DEBG|08:48:43.861]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10786186720),

```

dariuszseweryn, 2020-08-26:
Hello,
What happens if you postpone the first write to the characteristic by about 500 ms?

velthune, 2020-08-26:
Hi, thanks for answering.
I've just tried with 500, 1000, 2000, 10000ms and nothing happens unfortunately 😭

dariuszseweryn, 2020-08-26:
Could you share the updated log?

velthune, 2020-08-26:
Sure, I've setup verbose log mode:

```
2020-08-26 17:41:11.305090+0200 FooProject[1448:1194730] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x28278b4f0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
    [RxBLEKit|DEBG|15:41:11.314]: CentralManager(10778883312) didUpdateState(state: poweredOn)
[RxBLEKit|DEBG|15:41:11.398]: CentralManager(10778883312) retrievePeripherals(
    withIdentifiers: [1E26851E-3D5D-2834-92D2-B9C6D63AA4B5])
[RxBLEKit|DEBG|15:41:11.399]: CentralManager(10778883312) connect(
    peripheral: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")),
options: nil)
[RxBLEKit|DEBG|15:41:11.457]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverServices(
    serviceUUIDs: nil)
[RxBLEKit|DEBG|15:41:11.458]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverServices(services
: Optional("[Service(uuid: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E, id: 10764097088)]"),
    error: nil)
[RxBLEKit|DEBG|15:41:11.458]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverCharacteristics(
    characteristicUUIDs: nil,
for: Service(uuid: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E, id: 10764097088))
[RxBLEKit|DEBG|15:41:11.459]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverCharacteristicsFor(for:
Service(uuid: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E, id: 10764097088), characteristics:
    Optional("[Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779894976), Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779900736), Characteristic(uuid: 6E4000CA-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779890464), Characteristic(uuid: 6E40CCC0-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779886912), Characteristic(uuid: 6E40CCC1-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779901696)]"),
        error: nil)
[RxBLEKit|DEBG|15:41:11.460]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
for: Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779894976))
[RxBLEKit|DEBG|15:41:11.460]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
for: Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779900736))
[RxBLEKit|DEBG|15:41:11.460]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
for: Characteristic(uuid: 6E4000CA-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779890464))
[RxBLEKit|DEBG|15:41:11.460]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
for: Characteristic(uuid: 6E40CCC0-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779886912))
[RxBLEKit|DEBG|15:41:11.460]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) discoverDescriptors(
for: Characteristic(uuid: 6E40CCC1-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779901696))
[RxBLEKit|DEBG|15:41:11.461]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779894976), descriptors:
    Optional("[]"),
        error: nil)
[RxBLEKit|DEBG|15:41:11.461]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779900736), descriptors:
    Optional("[Descriptor(uuid: Client Characteristic Configuration, id: 10739763264)]"),
        error: nil)
[RxBLEKit|DEBG|15:41:11.461]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 6E4000CA-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779890464), descriptors:
    Optional("[]"),
        error: nil)
[RxBLEKit|DEBG|15:41:11.462]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 6E40CCC0-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779886912), descriptors:
    Optional("[]"),
        error: nil)
[RxBLEKit|DEBG|15:41:11.462]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didDiscoverDescriptorsFor
(for:Characteristic(uuid: 6E40CCC1-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779901696), descriptors:
    Optional("[]"),
        error: nil)
2020-08-26 17:41:11.485668+0200 FooProject[1448:1188901] [javascript] [FooProject] Setup MONITOR listener RX
    [RxBLEKit|DEBG|15:41:11.495]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) setNotifyValue(
    enabled: true,
for: Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779900736))
[RxBLEKit|DEBG|15:41:11.540]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didUpdateNotificationStateFor(
for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779900736), isNotifying: true,
    error: nil)
2020-08-26 17:41:13.499798+0200 FooProject[1448:1188901] [javascript] [FooProject] SEND command
    [RxBLEKit|DEBG|15:41:13.504]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) writeValue(
[... cut ...]
    [RxBLEKit|DEBG|15:41:18.613]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) writeValue(
    data: 41542b5245424f4f540d0a,
for: Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779894976),
type: withResponse)
[RxBLEKit|DEBG|15:41:18.681]: Peripheral(uuid: 1E26851E-3D5D-2834-92D2-B9C6D63AA4B5, name: Optional("ffffff1000016291")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779894976),
value: nil,
    error: nil)
2020-08-26 17:41:18.686064+0200 FooProject[1448:1188901] [javascript] [FooProject] SEND command resolve
```

dariuszseweryn, 2020-08-26:
I see nothing indicating that the iPhone did receive any data via notifications. I am afraid that you need to sniff the radio traffic.
Notifications seem to be correctly set up and writes are successful.

dariuszseweryn, 2020-08-27:
Could you show an equivalent log from an Android device? Maybe from comparison something interesting will emerge?

IaroslavTernovy, 2020-08-28:
This are logs from Android device
```
2020-08-28 09:45:29.166 9162-26117/com.urbanasolutions.toolkit D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=C8:95:D7:2A:DA:57
2020-08-28 09:45:29.167 9162-26117/com.urbanasolutions.toolkit D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2020-08-28 09:45:29.168 9162-26117/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2020-08-28 09:45:29.169 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2020-08-28 09:45:29.169 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2020-08-28 09:45:29.172 9162-26279/com.urbanasolutions.toolkit D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(165008316) in 1791 ms
2020-08-28 09:45:29.174 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->getLongVolatile(Ljava/lang/Object;J)J (greylist, linking, allowed)
2020-08-28 09:45:29.252 9162-26191/com.urbanasolutions.toolkit D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(109338043)
2020-08-28 09:45:29.253 9162-26280/com.urbanasolutions.toolkit D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(109338043)
2020-08-28 09:45:29.254 9162-26283/com.urbanasolutions.toolkit D/BluetoothGatt: discoverServices() - device: C8:95:D7:2A:DA:57
2020-08-28 09:45:29.606 9162-26117/com.urbanasolutions.toolkit D/BluetoothGatt: onConnectionUpdated() - Device=C8:95:D7:2A:DA:57 interval=6 latency=0 timeout=500 status=0
2020-08-28 09:45:29.786 9162-26117/com.urbanasolutions.toolkit D/BluetoothGatt: onSearchComplete() = Device=C8:95:D7:2A:DA:57 Status=0
2020-08-28 09:45:29.787 9162-26117/com.urbanasolutions.toolkit D/RxBle#BluetoothGatt$1: onServicesDiscovered status=0
2020-08-28 09:45:29.788 9162-26117/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2020-08-28 09:45:29.788 9162-26117/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2020-08-28 09:45:29.789 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2020-08-28 09:45:29.789 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2020-08-28 09:45:29.789 9162-26282/com.urbanasolutions.toolkit V/RxBle#ServiceDiscoveryOperation: Preparing services description
2020-08-28 09:45:29.792 9162-26282/com.urbanasolutions.toolkit V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    PERIPHERAL ADDRESS: C8:95:D7:2A:DA:57
    PERIPHERAL NAME: ffffff1000015fd8
    -------------------------------------------------------------------------
    
    Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
    Instance ID: 1
    -> Characteristics:
    	* Device Name (00002a00-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ WRITE ]
    	* Appearance (00002a01-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    	* Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    	* Central Address Resolution (00002aa6-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    
    Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
    Instance ID: 10
    -> Characteristics:
    
    Primary Service - Unknown service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
    Instance ID: 11
    -> Characteristics:
    	* Unknown characteristic (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ WRITE_NO_RESPONSE WRITE ]
    	* Unknown characteristic (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    	* Unknown characteristic (6e4000ca-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ WRITE_NO_RESPONSE WRITE ]
    	* Unknown characteristic (6e40ccc0-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ WRITE_NO_RESPONSE WRITE ]
    	* Unknown characteristic (6e40ccc1-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ WRITE_NO_RESPONSE WRITE ]
    --------------- ====== Finished peripheral content ====== ---------------
2020-08-28 09:45:29.794 9162-26280/com.urbanasolutions.toolkit D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(109338043) in 541 ms
2020-08-28 09:45:29.795 9162-26282/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->getLongVolatile(Ljava/lang/Object;J)J (greylist, linking, allowed)
2020-08-28 09:45:29.870 9162-26191/com.urbanasolutions.toolkit D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
[... cut ...]
2020-08-28 09:45:32.556 9162-26117/com.urbanasolutions.toolkit D/RxBle#BluetoothGatt$1: onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e
2020-08-28 09:45:32.556 9162-26117/com.urbanasolutions.toolkit W/lutions.toolki: Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2020-08-28 09:45:32.558 9162-26117/com.urbanasolutions.toolkit D/RxBle#BluetoothGatt$1: onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e
```

velthune, 2020-08-31:
@dariuszseweryn we found the issue:

we sent an "hex" command that was interpreted as string, but we need to send command as byte array

dariuszseweryn, 2020-08-31:
Nothing that grabbed my sight. From my experience what you observe is not an Android/iOS difference but something different. I would look at the peripherals you talk to — maybe they have different firmwares? Or maybe one of them is not "paired" (with the particular central in a proprietary way) and/or bonded somehow? Not sure about the flow they should have. Personally I have never seen a problem of this kind when using different platforms.

dariuszseweryn, 2020-09-02:
So I think I can safely close this issue.

closed 2020-09-02 by dariuszseweryn