dotintent/react-native-ble-plx #734: [Android] Connect and discover services - BleError: Device 00:16:A4:4C:B9:6A was disconnected

Sumukha1791, 2020-08-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. 

- [ ] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

Should be able to connect and discover services

## Current Behavior

Receiving a disconnection error with status code 61 when trying to connect and discover services for a BLE device using an Android phone. The way I have designed my code is that when I receive an error in connection request from react-native-ble-plx library I again try to connect to the BLE device in a loop for 5 attempts, if all the 5 attempts result in error on 6th attempt I pass refershGatt param with value onconnected to the connect call. This does clear the cache and I am able to connect to the device but the service descriptions are empty. The device I try to connect uses Bluetooth dongle so the dongle might be paired to the Android phone in a previous attempt so when I manually unpair/forget the Bluetooth dongle from the paired devices list it does successfully discover the services. I am expecting to connect to the BLE device without manually unpairing the BLE dongle from the paired devices list in the android phone.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Connect to a device using Bluetooth dongle
```js
bleManager.connectToDevice(deviceId)
```
2. Then discover the services and characteristics
```js
bleManager.discoverAllServicesAndCharacteristicsForDevice(deviceId)
```
3.  Disconnect from device
```js
bleManager.cancelDeviceConnection(deviceId)
```
4. Connect and try to discover services to a different device using the same Bluetooth dongle but pass refreshgatt parameter with the connect request.
```js
bleManager.connectToDevice(deviceId, { refreshGatt: 'OnConnected' })
```

### Context

* Library version: 1.0.3
* Platform: Android
* Platform logs (logcat):
```logs
2020-08-28 17:00:46.338 4052-4719/com.arubacx I/RxBle#RefreshGattCustomOperation: Calling BluetoothGatt.refresh() status: Success
2020-08-28 17:00:46.730 4052-4411/com.arubacx D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=61
2020-08-28 17:00:46.733 4052-4411/com.arubacx I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (00:16:A4:4C:B9:6A)
2020-08-28 17:00:46.740 4052-4733/com.arubacx W/RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
    java.lang.InterruptedException
        at java.lang.Object.wait(Native Method)
        at com.polidea.rxandroidble.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
        at com.polidea.rxandroidble.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:73)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:761)
2020-08-28 17:00:46.763 4052-4733/com.arubacx D/RxBle#ConnectionOperationQueue: FINISHED (13710691) in 462 ms
2020-08-28 17:00:46.763 4052-4733/com.arubacx D/RxBle#Executors$RunnableAdapter: Terminated.
2020-08-28 17:00:46.769 4052-4734/com.arubacx D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(239483829)
2020-08-28 17:00:46.770 4052-4713/com.arubacx D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(239483829)
2020-08-28 17:00:46.776 4052-4713/com.arubacx D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(239483829) in 6 ms
2020-08-28 17:00:53.558 4052-4734/com.arubacx D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(258339723)
2020-08-28 17:00:53.559 4052-4713/com.arubacx D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(258339723)
2020-08-28 17:00:53.562 4052-4719/com.arubacx V/RxBle#BleConnectionCompat: Connecting without reflection
2020-08-28 17:00:56.828 4052-4411/com.arubacx D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
2020-08-28 17:00:56.834 4052-4713/com.arubacx D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(258339723) in 3273 ms
2020-08-28 17:00:56.895 4052-4431/com.arubacx D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(109802079)
2020-08-28 17:00:56.897 4052-4733/com.arubacx D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(109802079)
2020-08-28 17:00:56.931 4052-4065/com.arubacx D/RxBle#BluetoothGatt: onServicesDiscovered status=0
2020-08-28 17:00:56.933 4052-4734/com.arubacx V/RxBle#ServiceDiscoveryOperation: Preparing services description
2020-08-28 17:00:56.937 4052-4734/com.arubacx V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    PERIPHERAL ADDRESS: 00:16:A4:4C:B9:6A
    PERIPHERAL NAME: JL668A---------
    -------------------------------------------------------------------------
    --------------- ====== Finished peripheral content ====== ---------------
2020-08-28 17:00:56.942 4052-4733/com.arubacx D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(109802079) in 42 ms## 
```

closed 2020-11-09 by stale[bot]

jnilesh, 2021-07-26:
while trying to connect to the device, I too get the same error multipleTimes

nolife08021, 2022-04-18:
@Sumukha1791 @jnilesh  hi did u guys found any solution for this? I'm facing the same issue.

Hassaan68, 2022-06-21:
@Sumukha1791 did you find any solution for this issue?

Tadejo9, 2022-06-21:
@Hassaan68 what versions of react native, library and os are you using?

Tadejo9, 2022-06-21:
I'm quoting this from official documentation of this library. 

> To connect to a device one needs to first [scan it](https://github.com/Polidea/react-native-ble-plx/wiki/Bluetooth-Scanning).

And this is the url of the "Connect" page of documentation https://github.com/dotintent/react-native-ble-plx/wiki/Device-Connecting

Hassaan68, 2022-06-21:
@Tadejo9 . react native version 0.66.4. Device is iphone 12 (ios 15.4.1)

Tadejo9, 2022-06-21:
@Hassaan68 have you tried scanning for devices?

Hassaan68, 2022-06-21:
@Tadejo9 the error occurs when update the device firmware, remove device from bluetooth pairing and then tries to reconnect to it. It just does not reconnect in first try and after the error is shown when I try to connect it then it works fine as before.

Tadejo9, 2022-06-21:
The error above is an Android error. Can you paste yours below?

Hassaan68, 2022-06-22:
@Tadejo9 . one more update. If I disconnect the device and remove from bluetooth pairing and then reconnects the device without firmware update then it works fine. It is worth noticing that firmware update changes the few characteristics of the device. It can be case that library is not aware of these changes and tries to reconnect to device by using the previous characteristics ?