dotintent/react-native-ble-plx #736: Android/ ConnectionOptions.requestMTU and connectToDevice.mtu is null?

juhanilammassaari, 2020-09-03:
I was using older react-native-ble-plx 1.1.1 and it has been working like a charm, but needed to upgrade react-native to latest 0.63.2 and then react-native-ble-plx 2.0.1 as older ble library would not compile.

Seems that almost everything worked just fine after upgrade, but I am struggling with 'mtu' field of the device object after connection. I am using { autoConnect: false, requestMTU: 247 } as ConnectionOptions, when connecting to a ble device using my phone.
With previous version the returned device.mtu was actually 247 after connected, but with new library this field is left null. I am using the same phone (Samsung S8) and BLE peripheral device so the setup is fixed.

**Code snippet:**
```
const deviceId = [my device id resolving with ble scan]
const options = {
  autoConnect: false,
  requestMTU: 247
};
console.log("Attempting to connect " + deviceId);
const connectedDevice = await this.bleManager.connectToDevice(deviceId, options);
console.log("Connected device MTU: " + connectedDevice.mtu);
```

**In the logs I can see (I have trimmed the log a bit)**

```
2020-09-03 03:11:38.328 31324-31424/my.app I/ReactNativeJS: Attempting to connect XX:XX:XX:XX:XX:XX 
2020-09-03 03:11:38.366 31324-31468/my.app D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(261493641)
2020-09-03 03:11:38.369 31324-31447/my.app D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(261493641)
...
2020-09-03 03:11:39.112 11718-11858/? E/bt_stack: [ERROR:gatt_sr.cc(752)] MTU request PDU with MTU size 247
2020-09-03 03:11:39.112 11718-11858/? E/bt_btm: BTM_SetBleDataLength failed, peer does not support request
...
2020-09-03 03:11:39.186 31324-31447/my.app D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(261493641) in 818 ms
...
2020-09-03 03:11:39.223 31324-31468/my.app D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(167280918)
2020-09-03 03:11:39.225 31324-31467/my.app D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(167280918)
2020-09-03 03:11:39.232 31324-31463/my.app D/BluetoothGatt: configureMTU() - device: XX:XX:XX:XX:XX:XX mtu: 247
...
2020-09-03 03:11:39.599 31324-31361/my.appl D/BluetoothGatt: onConfigureMTU() - Device=XX:XX:XX:XX:XX:XX mtu=247 status=0
2020-09-03 03:11:39.602 31324-31361/my.appl D/RxBle#BluetoothGatt$1: onMtuChanged mtu=247 status=0
2020-09-03 03:11:39.620 31324-31467/my.appl D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(167280918) in 393 ms
2020-09-03 03:11:39.622 31324-31424/my.appl I/ReactNativeJS: Connected device MTU: null
```

I am wonder whether this field is actually changing in native side, or is the connectToDevice() returning the connected device as soon ConnectOperation is finished. This might leave mtu not being set and even if in native side the configureMTU is called after connect, the react-native side would receive device object without mtu field being set (not having even the default value 23)? It could make sense, though changes functionality from 1.1.1 a bit.

Would you recommend me to call requestMTUForDevice() for the connected device once more even if I had the desired mtu in ConnectionOptions.

Thank you for the great library and support in advance!

Datagrdn, 2020-10-08:
I am having this same issue. Does anyone have a working of example of negotiating MTU on Android?

dariuszseweryn, 2020-12-16:
Hello,

We are looking into this. Temporary workaround would be to not set MTU in connection request options but use `.requestMTU()` right after the connection is established.

closed 2021-06-08 by dariuszseweryn