dotintent/react-native-ble-plx #738: startDeviceScan error - "Cannot start scanning operation" on Android (Samsung Galaxy only)

paulclark94, 2020-09-05:
I have built a RN app using this library. It works fine on the Galaxy Tab A I develop on but for some reason on my Samsung Galaxy S9+ the startDeviceScan method returns the following error:
`{"message":"Cannot start scanning operation","errorCode":600,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Scan failed because application registration failed (code 6)","name":"BleError","line":24096,"column":28,"sourceURL":"http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"}`

I've reviewed all comments on the required permissions etc. I'm using RN 0.62, build versions below:
`ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 21
        compileSdkVersion = 29
        targetSdkVersion = 29
        androidXCore = "1.0.2"
    }`

The permissions in AndroidManfiest.xml are:
`    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission `android:name="android.permission.FOREGROUND_SERVICE"/>`

And I have also manually requested as well through: 
`const granted = await PermissionsAndroid.request(
            PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
            {
                title: 'Permission Localisation Bluetooth',
                message: 'Requirement for Bluetooth',
                buttonNegative: 'Cancel',
                buttonPositive: 'OK',
            }
        );`

I have basically created a fresh basic app to test and am only calling the startDeviceScan to recreate this error. Like I mentioned before, the tablet works perfectly in all of the above situations, it is only my s(+ mobile that is not throwing the error.

closed 2020-11-14 by stale[bot]

pixelfoundry-dev, 2020-12-11:
@paulclark94 hi, i get the same error here, it worked for some minutes, then i cannot make it working again. This is really strange issue. Have you resolved it?

salmazov, 2021-04-06:
Have the same issue

semihaydintepe, 2021-09-09:
I guess Samsung phones and tablets have similar problems. I am getting the same error on Samsung Galaxy A02 and A31. I even had similar problem with Huawei Y6s.

wildseansy, 2021-09-15:
I am getting this issue on an android pixel 3XL

jegor377, 2023-07-08:
I'm having the same issue on my Samsung device. I've been trying to fix it for several hours. Nothing is working.