dotintent/react-native-ble-plx #743: Background Mode Documentation

captDaylight, 2020-09-10:
Under the instructions for iOS background mode there is:

> Pass `restoreStateIdentifier` and `restoreStateFunction` to `BleManager` constructor.

I think this can be elaborated. Can `restoreStateIdentifier` be any string? What's the motivation for `restoreStateFunction` and how should it be used?

I think even a init template in the `readme` would provide a lot of context to help future users out.

```ts
const bleManager = new BleManager({
  restoreStateIdentifier: 'mystring',
  restoreStateFunction: (bleRestoredState: BleRestoredState) => {
     // ??
  },
});
```

LucasAndrad, 2020-09-24:
Yes, I believe the docs should also include a real example and details about the `BleManager` state inside of `restoreStateFunction`, do we need a new instance of `BleManager`? Do we need to reconnect with the device(s)?

closed 2020-12-04 by stale[bot]

ortonomy, 2021-05-14:
Not sure this is stale. I want to know what this is and how to use it properly

LucasAndrad, 2021-05-14:
@ortonomy  Apple will kill your app after a few minutes (maybe a couple of hours) if the iphone peformance starts to leeking. The restore feature will restore your app and it's previous bluetooth status.

MaximeLozach, 2022-04-25:
Can someone give an example ?
I'm currently struggling with BleManager making my connection in a state I can't programatically recover from.
It looks like setting BleManageroptions could resolve this, but I don't relly get how it works.
(my issue :  #969 )