dotintent/react-native-ble-plx #744: Bug Report when I log the device and I log the deviceConnected

vagnerwentz, 2020-09-11:
##  Expected Behavior

I would like to print the `serviceUUIDs` when I connect to BlueTooth Module 

## Current Behavior

Inside my function called `scanAndConnect()` I call the function `manager.startDeviceScan()`, but the "bug" is about inside my 
`if (device.name === 'MLT-BT05' || device.id === '88:25:83:F0:30:BC')`. The strange behavior is when I `console.log(device)` this returns me this
[image]
 
When I use `const deviceConnected = await device.connect()` when I log `console.log(deviceConnected)` returns this
[image]

If we compare the two images we can see that some properties at `deviceConnected` is `null` and I would like to know why, because some functions at library needs the `servicesUUIDs`

# Function
```
async function scanAndConnect() {
    manager.startDeviceScan(null, null, async (error, device) => {
      if (error) {
        console.log('Error at if(error): ', error);
        return null;
      }

      if (!device) {
        console.log('Error at if(device === null)');
        return null;
      }

      if (device.name === 'MLT-BT05' || device.id === '88:25:83:F0:30:BC') {
        manager.stopDeviceScan();

        console.log(device);

        /* serviceUUID = 0000ffe0-0000-1000-8000-00805f9b34fb */
        if (device.serviceUUIDs) {
          console.log(device.serviceUUIDs);
          console.log(device.name);
        }

        const deviceConnected = await device.connect();

        console.log(deviceConnected);

        console.log(deviceConnected.id);
        console.log(deviceConnected.name);
        console.log(deviceConnected.serviceUUIDs); /* HERE THE servicesUUIDs is NULL */
      }
    });
  }
```

wvanooijen92, 2020-09-18:
`manager.stopDeviceScan();` looks a bit sketchy inside the callback of `startDeviceScan`

My approach is to scan for a list of devices for x seconds.
and put that in a state
```

  const toggleScanDevices = () => {
    setIsScanning(true);
    setBleDevices([]);

    bleManager.startDeviceScan(null, {}, (bleError, device) => {
      // note devices may be scanned multiple times
      if (device && _.findIndex(bleDevices, {id: device.id}) < 0) {
        bleDevices.push(device);
        setBleDevices(bleDevices); // set it in the state, we will use it later
      }
    });

    setTimeout(() => {
      setIsScanning(false);
      bleManager.stopDeviceScan();
    }, 5000);

```

Check if the states knows about the device, and if so, connect to it
```
  // let's call this with 'MLT-BT05'
  const connectDevice = async (name) => {
    let device = findDeviceWhereNameContains(name); // get's device object from state by name

    if (device == null) {
      setConnectedDevices({...connectedDevices, [name]: null});
      return false;
    }

    let isConnected = await device.isConnected();

    if (!isConnected) {
      device = await bleManager.connectToDevice(device.id);
      isConnected = await device.isConnected();
    }
    // console.log(isConnected, device);
    device = await device.discoverAllServicesAndCharacteristics();

    device.onDisconnected((error, device) => {
      setConnectedDevices({...connectedDevices, [name]: null});
    });

    const services = await device.services();
    const characteristics = {};
    const descriptors = {};
    await _.forEach(services, async (service) => {
      characteristics[service.uuid] = await device.characteristicsForService(
        service.uuid,
      );
    });

    setConnectedDevices({
      ...connectedDevices,
      [name]: {
        connected: isConnected,
        services,
        device,
        characteristics,
      },
    });
  };

```

vagnerwentz, 2020-09-18:
> `manager.stopDeviceScan();` looks a bit sketchy inside the callback of `startDeviceScan`
> 
> My approach is to scan for a list of devices for x seconds.
> and put that in a state
> 
> ```
> 
>   const toggleScanDevices = () => {
>     setIsScanning(true);
>     setBleDevices([]);
> 
>     bleManager.startDeviceScan(null, {}, (bleError, device) => {
>       // note devices may be scanned multiple times
>       if (device && _.findIndex(bleDevices, {id: device.id}) < 0) {
>         bleDevices.push(device);
>         setBleDevices(bleDevices); // set it in the state, we will use it later
>       }
>     });
> 
>     setTimeout(() => {
>       setIsScanning(false);
>       bleManager.stopDeviceScan();
>     }, 5000);
> ```
> 
> Check if the states knows about the device, and if so, connect to it
> 
> ```
>   // let's call this with 'MLT-BT05'
>   const connectDevice = async (name) => {
>     let device = findDeviceWhereNameContains(name); // get's device object from state by name
> 
>     if (device == null) {
>       setConnectedDevices({...connectedDevices, [name]: null});
>       return false;
>     }
> 
>     let isConnected = await device.isConnected();
> 
>     if (!isConnected) {
>       device = await bleManager.connectToDevice(device.id);
>       isConnected = await device.isConnected();
>     }
>     // console.log(isConnected, device);
>     device = await device.discoverAllServicesAndCharacteristics();
> 
>     device.onDisconnected((error, device) => {
>       setConnectedDevices({...connectedDevices, [name]: null});
>     });
> 
>     const services = await device.services();
>     const characteristics = {};
>     const descriptors = {};
>     await _.forEach(services, async (service) => {
>       characteristics[service.uuid] = await device.characteristicsForService(
>         service.uuid,
>       );
>     });
> 
>     setConnectedDevices({
>       ...connectedDevices,
>       [name]: {
>         connected: isConnected,
>         services,
>         device,
>         characteristics,
>       },
>     });
>   };
> ```
> 
> I hope it helps

Do you have some repository to I check and see?

wvanooijen92, 2020-09-18:
No sorry, this project is not open source. 

But implementing those two things should not be to complicated, it's mainly plain react.

My original post, has the complete file. 
https://github.com/Polidea/react-native-ble-plx/issues/747

vagnerwentz, 2020-09-18:
> No sorry, this project is not open source.
> 
> But implementing those two things should not be too complicated, it's mainly plain react.
> 
> My original post, has the complete file.
> #747

Ok I will try your tip, and if works, I say to you.

vagnerwentz, 2020-09-18:
> No sorry, this project is not open source.
> 
> But implementing those two things should not be too complicated, it's mainly plain react.
> 
> My original post, has the complete file.
> #747

Hey friend, I saw that you created some functions out the component, can you explain to me about this?

vagnerwentz, 2020-09-18:
> They are simply helper functions, the functions outside the components do not depend on what happens within the component, so I don't want them to be created for every re-render.

I did not understand this part
```
const toggleScanDevices = () => {
    setIsScanning(true);
    setBleDevices([]);

    bleManager.startDeviceScan(null, {}, (bleError, device) => {
      // note devices may be scanned multiple times
      if (device && _.findIndex(bleDevices, {id: device.id}) < 0) {
        bleDevices.push(device);
        setBleDevices(bleDevices); // set it in the state, we will use it later
      }
    });

    setTimeout(() => {
      setIsScanning(false);
      bleManager.stopDeviceScan();
    }, 5000);
```
Where contains the `_` at the `if(...)`, here does not work

wvanooijen92, 2020-09-18:
that `_` is coming from the `lodash` library

vagnerwentz, 2020-09-18:
> that `_` is coming from the `lodash` library

About your state `connectedDevices, setConnectedDevices` I saw that you use an object with three variables inside, how can I know about these variables?

wvanooijen92, 2020-09-18:
Yeah in my app I need to connect to three devices at the same time, so what you see is that I store the device by device name in that object.

vagnerwentz, 2020-09-18:
> Yeah in my app I need to connect to three devices at the same time, so what you see is that I store the device by device name in that object.

Ah understood, I will need to connect to just one device, the map will (cellphone - module bluetooth), so what I can put at the state?

wvanooijen92, 2020-09-18:
I guess you can just store the device?

wvanooijen92, 2020-09-18:
I just looked back at your initial question, are you sure that you cache the characteristics and services with `discoverAllServicesAndCharacteristics`

vagnerwentz, 2020-09-18:
> I just looked back at your initial question, are you sure that you cache the characteristics and services with `discoverAllServicesAndCharacteristics`

No

vagnerwentz, 2020-09-18:
> I just looked back at your initial question, are you sure that you cache the characteristics and services with `discoverAllServicesAndCharacteristics`

I do not understand what is this services and characteristics, I tried to understand seeing the doc, but not worked.

wvanooijen92, 2020-09-18:
https://lmgtfy.app/?q=ble+gatt+services+and+characteristics

vagnerwentz, 2020-09-18:
> https://lmgtfy.app/?q=ble+gatt+services+and+characteristics

One doubt bro, the `const const [bleDevices, setBleDevices] = useState([]);` will the all devices that the bluetooth can see? Regardless if I'm going to use them or not?

vagnerwentz, 2020-09-25:
> https://lmgtfy.app/?q=ble+gatt+services+and+characteristics

Hey, how are you? I implemented using your code and change somethings but occurs the same when I connect to the device, and I log the device, return some null values.

closed 2020-10-06 by vagnerwentz

JamesOwen7, 2022-04-23:
@vagnerwentz did you solve this issue?