dotintent/react-native-ble-plx #747: Subscribing to two notifiable characteristics simultaneously

wvanooijen92, 2020-09-16:
Great package, was able to get some ideas going super quickly. Though I ran into a problem; Is it possible to subscribe to two notifiable characteristics at the same time?

I am working on a POC where we want to measure multiple data points simultaneously. Two of the three devices use a notifiable characteristic. I notice that subscribing to a notification stream and reading from a readable characteristic is possible. But when I try to listen to both streams things just kinda freeze. 

I am going to spend more time on researching if this is me, not being able to write react. Though, if there are any limitations with regard to simultaneous subscriptions on multiple devices, it would be nice to find out. 

This is my code

```
import React, {useEffect, useState} from 'react';
import {
  SafeAreaView,
  StyleSheet,
  ScrollView,
  View,
  Text,
  StatusBar,
  PermissionsAndroid,
} from 'react-native';
import _ from 'lodash';
import {Colors} from 'react-native/Libraries/NewAppScreen';
import {BleManager} from 'react-native-ble-plx';
import {
  DefaultTheme,
  Button,
  List,
  Provider as PaperProvider,
  Chip,
} from 'react-native-paper';

import Grid, {Row, Col} from './Grid/Grid';

const getCharacteristicContainingUuids = (
  serviceUuid,
  characteristicUuid,
  characteristics,
) => {
  // console.log(
  //   serviceUuid,
  //   characteristics[serviceUuid],
  //   _.find(characteristics[serviceUuid], {uuid: characteristicUuid}),
  //   characteristics,
  // );

  return _.find(characteristics[serviceUuid], {uuid: characteristicUuid});
};

const CharacteristicData = ({
  name,
  serviceUuid,
  characteristicUuid,
  bleState,
}) => {
  const [value, setValue] = useState(null);
  const [subscription, setSubscription] = useState(null);

  const characteristic = getCharacteristicContainingUuids(
    serviceUuid,
    characteristicUuid,
    bleState.characteristics,
  );

  // console.log(characteristic);

  useEffect(() => {
    return () => {
      if (subscription) {
        subscription.remove();
        setSubscription(null);
      }
[... 380 lines cut ...]
    color: Colors.black,
  },
  sectionDescription: {
    marginTop: 8,
    fontSize: 18,
    fontWeight: '400',
    color: Colors.dark,
  },
  highlight: {
    fontWeight: '700',
  },
  footer: {
    color: Colors.dark,
    fontSize: 12,
    fontWeight: '600',
    padding: 4,
    paddingRight: 12,
    textAlign: 'right',
  },
});

export default App;
```

I hope anyone can help me! Thank you

vagnerwentz, 2020-09-17:
Hello. My comment does not have a sense about your question, but I would like to know if you can connect your cellphone to Bluetooth, because my application I can connect but returns me some null values.

wvanooijen92, 2020-09-18:
I am not really sure what you mean. Connecting with BLE devices is no problem. When I subscribe to a second characteristic, both components seem to freeze. No errors.

vagnerwentz, 2020-09-18:
> I am not really sure what you mean. Connecting with BLE devices is no problem. When I subscribe to a second characteristic, both components seem to freeze. No errors.

I opened an issue if you can see to understand. What I did say. My problem is after I call the `connect()` function.
https://github.com/Polidea/react-native-ble-plx/issues/744

vagnerwentz, 2020-09-21:
> I am not really sure what you mean. Connecting with BLE devices is no problem. When I subscribe to a second characteristic, both components seem to freeze. No errors.

Hey bro, how can I check if is connected? And how can I send a text for example to see if appears at the terminal bluetooth? Can you help me?

vagnerwentz, 2020-09-21:
Hey, I did this function using your code
```ts
const connectDevice = async (name: string) => {
    let device = findDeviceWhereNameContains(name);

    console.log('Primeiro: ', device);

    if (device === null) {
      setConnectedDevice(INITIAL_DEVICE_STATE);
      return false;
    }

    let isConnected = await device.isConnected();

    if (!isConnected) {
      device = await bleManager.connectToDevice(device.id);
      isConnected = await device.isConnected();
    }
    device = await device.discoverAllServicesAndCharacteristics();

    device.onDisconnected((error, device) => {
      setConnectedDevice(INITIAL_DEVICE_STATE);
    });

    const services = await device.services();
    const characteristics: Record<string, Characteristic[]> = {};
    const descriptors = {};

    _.forEach(services, async service => {
      const deviceCharacteristics = await device?.characteristicsForService(
        service.uuid,
      );
      characteristics[service.uuid] = deviceCharacteristics || [];
    });

    console.log('Segundo: ', connectedDevice.device);

    setConnectedDevice(state => ({
      ...state,
      device,
    }));

    console.log('Terceiro', connectedDevice.device);
  };
```
The first log, print any datas with the UUID and etc..., but where the log Segundo, and Terceiro, appears null.

vagnerwentz, 2020-09-21:
We can talk at WhatsApp or anything? I think it is better to explain what the issue I have

vagnerwentz, 2020-10-13:
@wvanooijen92 Hey, can we talk more?

closed 2020-12-24 by stale[bot]

Cleversou1983, 2021-12-10:
Hey man, could you solve this issue, on subscribing in more than one notifiable characteristic? I'm facing the same problem here... Thanks!

wvanooijen92, 2021-12-10:
I did solve it but not with this library instead I went with https://github.com/innoveit/react-native-ble-manager

Cleversou1983, 2021-12-31:
> I did solve it but not with this library instead I went with https://github.com/innoveit/react-native-ble-manager

Thanks, man! I'll check this out!