dotintent/react-native-ble-plx #748: Problem in decoding Characteristic value.

mhkasif, 2020-09-21:
## Question

Please describe the question.
I've simulated BLE peripheral using bleno.
below I'm attaching the pictures which show the onReadRequest function to which some value is sent to the device.
[image]

what is want in return is "800400000000" this value which is being shown by lightblue app.

[image]

I'm getting a base64 decoded value which I tried a lot to convert back to the original value but yet unsuccessful.

kindly tell me how to resolve this problem.

mhkasif, 2020-09-22:
Issue resolved by myself.
thanks

closed 2020-09-22 by mhkasif

TwoTenPvP, 2020-11-21:
How did you resolve it? Doesn't seem documented anywhere

hungdev, 2021-01-06:
I have the same question, how did you resolve it?