dotintent/react-native-ble-plx #749: Bug Report - Duplicate Devices Scanned when `allowDuplicates:false`

objectiveSee, 2020-09-24:
## Prerequisites

## Expected Behavior

Setting `allowDuplicates:false` in `startDeviceScan` should not produce duplicate scanned devices oniOS

## Current Behavior

I am seeing duplicates!

### Steps to Reproduce

Example code (index.js) and log output shows that my Macbook was scanned multiple times

https://gist.github.com/objectiveSee/cb43186979902b404a22d52daff8b709

### Context

* Library version: ^2.0.1
* Platform: iOS
* Contents of the `package.json` file:
```
{
  "name": "bluetooth_pi_client",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "ios-device": "react-native run-ios --device",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "react": "16.13.1",
    "react-native": "0.63.2",
    "react-native-ble-plx": "^2.0.1"
  },
  "devDependencies": {
    "@babel/core": "^7.11.6",
    "@babel/runtime": "^7.11.2",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.3.0",
    "eslint": "^7.9.0",
    "jest": "^26.4.2",
    "metro-react-native-babel-preset": "^0.63.0",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  }
}

```

thertzelle, 2020-11-16:
Ooof! I thought this was just me but maybe you are having the same problem!

objectiveSee, 2020-11-23:
@thertzelle I never solved it, but let me know if you find anything!

mikolak, 2020-11-23:
This might provide an insight for you https://stackoverflow.com/a/11731243/6413439

closed 2021-06-04 by stale[bot]