dotintent/react-native-ble-plx #750: [IOS] Got freezed when writing characteristic data

wathui99, 2020-09-26:
## Current Behavior

(IOS) Program was freezed when I tried to use writeWithoutResponse at second time. At the first time use writeWithoutResponse I didn't receive any data.
Note: characteristic reading works fine (IOS), and everything works perfectly when i run on Android.

### Steps

1. Scaning devices to get device ID
2. Connecting to device. //bleManager.connectToDevice(deviceID, {autoConnect: true,requestMTU: 187});
3. discoverAllServicesAndCharacteristics
4. Getting services from device
5. Getting characteristics from service
6. Using characteristics for reading and writing data

### Context

* Library version: 2.0.1
* RN: 0.61.5
* Platform: IOS
* Version: 13.4.1
* Device: Iphone 8 plus

```js
try {
  console.log("start sending 1");
  await characterTx.writeWithoutResponse(btoa("test 1"));
  console.log("start sending 2");
  await characterTx.writeWithoutResponse(btoa("test 2"));
  console.log("start sending 3");
  await characterTx.writeWithoutResponse(btoa("test 3"));
}
catch(err) {
  console.log(err);
}

LOG: start sending 1
LOG: start sending 2 (freezed here)
```

vagnerwentz, 2020-10-02:
Hello I have an issue but anyone could help me, can you help me? It is about after I connect to device, the device returns some null values, and these values are important to read or write.

FazilMuhammed, 2020-10-07:
@wathui99 did you find the solution.i would like to print image please help me??

vagnerwentz, 2020-10-07:
> @wathui99 did you find the solution.i would like to print image please help me??

Yeah I found the solution, but now I need the characteristic UUID but I do not know hot to get.

FazilMuhammed, 2020-10-07:
@vagnerwentz  Thank you for your replay.can you share your code please 

Here is my code  i thinks its will help you to find the characteristic

 scanAndConnect() {
        console.log("Scanning Started");
        this.manager.startDeviceScan(null, null, (error, device) => {
            if (error) {
                // Handle error (scanning will be stopped automatically)
                console.log("Error in scanning devices:", error);
                return
            }
            // Check if it is a device you are looking for based on advertisement data
            // or other criteria.
            console.log("Detected Device Details:", device.id, device.name);
            // ||device.localName === 'BLEPeripheralApp') 
            if (device.name === 'BlueTooth Printer') { //
                // Stop scanning as it's not necessary if you are scanning for one device.
                console.log("Device Found, Stopping the Scan.");
                console.log("Connecting to:", device.name)
                this.manager.stopDeviceScan();
                device.connect()
                    .then((device) => {                  
                        console.log("Connected...Discovering services and characteristics");
                        return device.discoverAllServicesAndCharacteristics()
                    })
                    .then(async (device) => {
                        console.log('Services and characteristics discovered');
                        const services = await device.services()
                        const characteristics = {};
                        services.forEach(async service => {
                            const deviceCharacteristics = await device?.characteristicsForService(
                                service.uuid,
                            );
                            characteristics[service.uuid] = deviceCharacteristics || [];
                        })
                        console.log(characteristics, "characteristics");
                    })
                    .then(() => {
                        console.log("Hello...")
                    }, (error) => {
                        console.warn(error.message);

                    })
            }
        });
    }

and Document says that 

{
    "uuid": string,                         // Characteristic UUID
    "serviceUUID": string,                  // Service UUID which owns this characteristic
    "deviceUUID": string,                   // Device identifier which owns this characteristic
    "isReadable": boolean,                  // Is characteristic readable
    "isWriteableWithResponse": boolean,     // Is characteristic writeable when writing with response
    "isWriteableWithoutResponse": boolean,  // Is characteristic writeable when writing without response
    "isNotifiable": boolean,                // Is characteristic notifiable
    "isNotifying": boolean,                 // Current status of notification for this characteristic
    "isIndictable": boolean,                // Is characteristic indictable
    "value": string,                        // Base64 value, may be null

    // Utility functions
    // ...
}

uuid is real Characteristic UUID

its will helps you.
i need after the CharacteristicUUID got. how can send or write the data to printer please help me

vagnerwentz, 2020-10-07:
This is my function
```
const connectDevice = useCallback(
    async (name: string) => {
      let device = findDeviceWhereNameContains(name);

      console.log(device);

      if (device === null) {
        setConnectedDevice(undefined);
        return false;
      }

      const valuesAtTheDeviceVariable = pick(device, [
        'serviceUUIDs',
        'overflowServiceUUIDs',
        'solicitedServiceUUIDs',
        'localName',
        'isConnectable',
        'txPowerLevel',
        'serviceData',
        'mtu',
        'manufacturerData',
        'rssi',
      ]);

      let isConnected = await device.isConnected();

      if (!isConnected) {
        device = await bleManager.connectToDevice(device.id);
        isConnected = await device.isConnected();
      }
      device = await device.discoverAllServicesAndCharacteristics();

      device.onDisconnected((error, device) => {
        setConnectedDevice(undefined);
      });

      const services = await device.services();
      const characteristics: Record<string, Characteristic[]> = {};
      const descriptors = {};

      _.forEach(services, async service => {
        const deviceCharacteristics = await device?.characteristicsForService(
          service.uuid,
        );
        characteristics[service.uuid] = deviceCharacteristics || [];
      });

      const newDevice = {
        ...pick(device, ['id', 'name', 'cancelConnection', 'isConnected']),
        ...valuesAtTheDeviceVariable,
      };
      setConnectedDevice(newDevice);
    },
    [findDeviceWhereNameContains],
  );
```

FazilMuhammed, 2020-10-08:
@vagnerwentz How send the data to Device. can you share your code completely the module how to scan, connect and write difficult to understand this much information.

FazilMuhammed, 2020-10-08:
@vagnerwentz Did you know how to print image or send image to printer

vagnerwentz, 2020-10-09:
@FazilMuhammed Hey when I use `device = await device.discoverAllServicesAndCharacteristics();` the return comes
```
Device {overflowServiceUUIDs: null, solicitedServiceUUIDs: null, localName: null, isConnectable: null, txPowerLevel: null, …}
overflowServiceUUIDs: null
solicitedServiceUUIDs: null
localName: null
isConnectable: null
txPowerLevel: null
serviceUUIDs: null
serviceData: null
mtu: null
name: "MLT-BT05"
manufacturerData: null
rssi: null
id: "88:25:83:F0:30:BC"
```

FazilMuhammed, 2020-10-12:
@vagnerwentz after the return device = await device.discoverAllServicesAndCharacteristics();
you should follow this line 

 getServicesAndCharacteristics(device) {
        return new Promise((resolve, reject) => {
            device.services().then(services => {
                const characteristics = []

                services.forEach((service, i) => {
                    service.characteristics().then(c => {
                        characteristics.push(c)

                        if (i === services.length - 1) {
                            const temp = characteristics.reduce(
                                (acc, current) => {
                                    return [...acc, ...current]
                                },
                                []
                            )
                            const dialog = temp.find(
                                characteristic =>
                                    characteristic.isWritableWithoutResponse
                            )
                            if (!dialog) {
                                reject('No writable characteristic')
                            }
                            resolve(dialog)
                        }
                    })
                })
            })
        })
    }

then you can access the characteristics

vagnerwentz, 2020-10-13:
@FazilMuhammed we can talk at whatsapp, linkedin or facebook? something else?

FazilMuhammed, 2020-10-19:
@vagnerwentz +919747559615 is my whatsapp number

closed 2020-12-25 by stale[bot]