dotintent/react-native-ble-plx #752: Question

oguilb, 2020-10-06:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [x ] I checked the documentation and found no answer
- [ x] I checked to make sure that this issue has not already been filed
- [ x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Hi

For several days, I have been trying to send 50 bytes to a device. But this one does not seem to receive them (because it does not answer)
I use the device.writeCharacteristicWithoutResponseForService () method 
I split my data into 20 bytes portions.
I receive correctly from device because, when I connect to this device it send data and my application must send some data.

Usually, I use the nordic library for android (Native developpment) and there with same data, same method (caracteristic , service and write without response). My device no respond correctly.

Now, I don't know what to do.

I saw that the lib ble-plx for Android not use the last version of rxBle lib. It's voluntary and an update can perhaps resolve my problem or not?

I split my data before to send and convert all portion to base64. It's ok for you? 

Best regards
Olivier

vagnerwentz, 2020-10-06:
One doubt that I have. How did you get the Characteristic's UUID?

oguilb, 2020-10-07:
HI,

I use yield call([device, device.discoverAllServicesAndCharacteristics]); to enumerate all services and characteristics. 
I know the UUID of the feature to use because I already use it in a native project with the Nordic Android lib and the device is made by my company.

I will still check that I was not wrong (it will be too stupid) :-)

Thanks

vagnerwentz, 2020-10-07:
> HI,
>
> I use yield call([device, device.discoverAllServicesAndCharacteristics]); to enumerate all services and characteristics.
> I know the UUID of the feature to use because I already use it in a [...]

Good, but I am not using Redux. ;(

oguilb, 2020-10-07:
I have verified. My UUID to write characteristic is correct. I'll test without redux if my schedule allows it.

closed 2020-10-16 by oguilb