dotintent/react-native-ble-plx #756: Bluetooth Scan and Connecting were fine.Getting Characteristics not get idea how its obtain ?? i need to print a Slip via Bluetooth printer

FazilMuhammed, 2020-10-08:
Couple months gets stuck to print a slip. try lot of methods not yet works anyone knows how its works please help me...

AmirHMousavi, 2020-10-09:
Actually, you need the manual/documentation of the printer to check the UUIDs of the  UART Rx and TX  and the UART service (which is known as characteristics and service ). After reading the printer technical documentation and discovering the characteristics and monitor the needed service and characteristics. you can read/write the value from/to the printer.

FazilMuhammed, 2020-10-12:
@AmirHMousavi Thanks your response.Now i understand how its obtain. i have doubt with this library is it possible to write image on Printer.i have a bluetooth printer i need print image if have any idea please help me .

AmirHMousavi, 2020-10-12:
This is related to your printer.  [Check the BLE PLX documentation for `writeCharacteristic`](https://polidea.github.io/react-native-ble-plx/#blemanagerwritecharacteristicwithresponsefordevice)
i.g:
```ts
bleManager.writeCharacteristicWithResponseForDevice(
        "printerID",
        "serviceUUID",
        "UART-UUID",
       convertedImageToBase64)
.the(result=>console.log(result))
```
The converted image probably should be divided into chunks regarding the MTU size that is negotiated at connection time. I am not completely sure, but this is what I get from documentations

closed 2020-12-24 by stale[bot]