dotintent/react-native-ble-plx #757: Causing the build to fail

tarunkishore2303, 2020-10-19:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Give me the successful build when I run `react-native run-android`

## Current Behavior

Causing my build to fail. And the build is successful when I uninstall the `react-native-ble-plx` package

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
    Install the latest version of react native and Android studio
2. step 2
    Install `react-native-ble-plx` and run the command `react-native run-android`
3. Step 3
    You will get errors in the terminal

When the `npm` package is installed:
[image]

Testing in : Android
Platform OS : Windows 10 Home

Mhde96, 2020-10-20:
i have the same problem

tarunkishore2303, 2020-11-15:
My bad ! Didn't change the `minSdkVersion` and those extra stuff !

closed 2020-11-15 by tarunkishore2303