dotintent/react-native-ble-plx #762: Aborting a connection while it is 'in flight'

joospotter, 2020-10-27:
Would there be a preferred way to handle aborting a connection request? 

The interface I am working on allows a user to connect to a device, but also to abort the connection mid process. Since I cannot find an abort function or similar, I just hide the connection state and wait in the background until the connection finishes, cancelling the connection at that point if it completes successfully.

This leads to some more complicated state juggling than I'd like and it would be simpler if the request could just be terminated mid flight. Is this possible to do?

mikolak, 2020-10-27:
You can just call `cancelConnection` before it finishes.
https://polidea.github.io/react-native-ble-plx/#devicecancelconnection

joospotter, 2020-10-27:
That seems to work well - thank you for your help.

closed 2020-10-27 by joospotter