dotintent/react-native-ble-plx #763: Receiving response from device not working

Tadejo9, 2020-11-04:
I'm trying to enable device specific functionality. To achieve that I need to write a command to characteristic with response. Due to response not being returned in returned characteristic I need to monitor that characteristic for response.

Ways of implementation I've tried and failed.

**Way one**
First I write command to the device. This goes smoothly. Then I setup Subscription to the same characteristic. After subscribing device is disconnected after 10 seconds. Latter happens before response from device could be retrieved.
```
const responseChar = await this.device.writeCharacteristicWithResponseForService(serviceUUID, charUUID, command)

this.device.monitorCharacteristicForService(serviceUUID, charUUID, (error, characteristics) => {
   if (error) {console.log(error)}
   if (characteristics) {console.log(characteristics.value)}
})
  
setTimeout(async () => {
  try {
     await this.device?.cancelConnection();
   } catch (error) {
     console.log(error)
   }
}, 10000)
```
**Way two**
I've read an [article](https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/) in which characteristicW and characteristicN are set with prefixUUID + xx + suffix.UUID. With this approach error is thrown while attempting to writeWithResponse with message that characteristic with this UUID does not exist.

How could i resolve problem described above?

Developing on MacOS.
Problem on Android and iOS.
Library version: 2.0.1

thomas-ot, 2020-11-05:
It looks like you need to subscribe to ATT Value Notification on the characteristic (with monitorCharacteristic...)  **before** writing to it.

Tadejo9, 2020-11-05:
This approach was also attempted, not successfully. I've also extended timeout to 30 seconds. Another possibility is that command is wrong, but in any case i should receive a response of Accept, Denied or In Progress.

closed 2020-12-03 by Tadejo9