dotintent/react-native-ble-plx #764: onStateChange fires events multiple times

tekatoka, 2020-11-06:
I am subscribing the phone bluetooth state with the onStateChange method of the BleManager:

```
        const subscription = bleManager.onStateChange((state) => {
            if (state === 'PoweredOn') {
                //scanAndConnect();
            } 

            if (state === 'PoweredOff') {
                //local notification
            } 
        }, true);
```
However, every time when the bluetooth state of the phone changes (I turn my bluetooth on or off), the event is firing multiple times. In my case, this causes up to 30 local notifications if the bluetooth is PoweredOff.

If I remove the subscription, the event is firing once but then never again until I reload the app. 

Expected behavior: getting response every time the Bluetooth state changes, but only _once_. 

How can this be achieved? I appreciate any help!

dariuszseweryn, 2020-12-05:
Are you sure that you do not subscribe multiple times to `.onStateChange()`?

weixiangyong, 2020-12-29:
我这边是正常的,你的手机是什么型号的

closed 2021-06-08 by dariuszseweryn