dotintent/react-native-ble-plx #769: TypeError: undefined is not an object (evaluating 'this.manager.startDeviceScan')

Girum, 2020-11-18:
Hi. I`m trying to use the lib, and write and run the introduction steps on the documentations. Already linked in react native, but always receaving this ,essage: TypeError: undefined is not an object (evaluating 'this.manager.startDeviceScan') when try to start Device Scan. I saw many issues and nothing had properly solutions. How can I Proceed?

mikolak, 2020-11-18:
I think you might have not assigned `this.manager`.

Girum, 2020-11-19:
No, I assigned then, like said this.manager in the constructor, but it not works as well. Can you have a simples snippet just to start and scan, just for begin?

dariuszseweryn, 2020-11-19:
Hello @Girum 
> Can you have a simples snippet just to start and scan, just for begin?

Isn't there a sample linked in the readme?

Girum, 2020-11-19:
> Hello @Girum
> 
> > Can you have a simples snippet just to start and scan, just for begin?
> 
> Isn't there a sample linked in the readme?

yes, there is, but it not works hahaha. Ever get the same problem...

`import { BleManager } from 'react-native-ble-plx';

constructor() {
    super();
    this.manager = new BleManager();
    ...
}

scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            return
        }

        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        if (device.name === 'TI BLE Sensor Tag' || 
            device.name === 'SensorTag') {
            
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();

            // Proceed with connection.
        }
    });
}`

With the sample above

dariuszseweryn, 2020-11-19:
What would be logged if you add a `console.log` in the constructor?
```
import { BleManager } from 'react-native-ble-plx';

constructor() {
    super();
    const manager = new BleManager();
    console.log('manager', manager);
    this.manager = manager;
    ...
}
```

dariuszseweryn, 2020-11-19:
And a bonus question — what device you try to run your code on?

Girum, 2020-11-19:
It takes this:

`{"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_subscriber": {"_currentSubscription": null, "_subscriptionsForType": [Object]}}, "_uniqueId": 0}`

I'm running on a physical device (Galaxy s10)

dariuszseweryn, 2020-11-19:
And what does `console.log(this.manager)` print just before you try to call `startDeviceScan()`?

Girum, 2020-11-19:
Says the same thing. Same output (printed in the constructor)

Girum, 2020-11-19:
+ I tried to permission the Bluetooth as well, using another library, and used here. It has the permission works for fine location (and too in the manifest)

dariuszseweryn, 2020-11-19:
So if `this.manager` is not an `undefined` (as you say the log is showing what you pasted) then `TypeError: undefined is not an object (evaluating 'this.manager.startDeviceScan')` has no sense.

Girum, 2020-11-19:
That`s the problem hahaha.

```
import React, { Component } from 'react';
import {
  View,
  Platform,
  PermissionsAndroid,
  Button,
} from 'react-native';

import { BleManager } from 'react-native-ble-plx';

export default class App extends Component {

  componentDidMount() {
    

    if (Platform.OS === 'android' && Platform.Version >= 23) {
        PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION).then((result) => {
            if (result) {
              console.log("Permission is OK");
            } else {
              PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION).then((result) => {
                if (result) {
                  console.log("User accept");
                } else {
                  console.log("User refuse");
                }
              });
            }
      });
    }

  }

  constructor() {
    super();
    const manager = new BleManager();
    console.log('manager', manager);
    this.manager = manager;
    console.log('manager.this', this.manager);
    
  }

  scanAndConnect() {
    console.log("entrou")
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            return
        }

        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        if (device.name === 'TI BLE Sensor Tag' || 
            device.name === 'SensorTag') {
            
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();

            // Proceed with connection.
        }
    });
}

render() {
  return(
    <View>
      <Button
        title="APERTE"
        onPress={this.scanAndConnect}
      />
    </View>
  );
}
}
```

That's my entire code

dariuszseweryn, 2020-11-19:
```
  scanAndConnect() {
    console.log("entrou", this.manager) // <<<<<<<<< change this line
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            return
        }

        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        if (device.name === 'TI BLE Sensor Tag' || 
            device.name === 'SensorTag') {
            
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();

            // Proceed with connection.
        }
    });
```

Girum, 2020-11-19:
Changed. It has the same erro (undefined is not an object (evaluating 'this.manager')

Girum, 2020-11-19:
I tested with a string, and it cannot access the variable, maybe this...

mikolak, 2020-11-19:
`onPress={this.scanAndConnect}` =>
```typescript
onPress={() => this.scanAndConnect()}
``

Girum, 2020-11-19:
> `onPress={this.scanAndConnect}` =>
> 
> ```ts
> onPress={() => this.scanAndConnect()}
> ```
> 
> https://github.com/microsoft/TypeScript/wiki/'this'-in-TypeScript
> 
> tl;dr `onPress` has different `this`

It stops the error, but, nothing happens. Now, if i Scan, it throws the error i paste first:

`{"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_subscriber": {"_currentSubscription": null, "_subscriptionsForType": [Object]}}, "_uniqueId": 0}`

Girum, 2020-11-19:
It prints ble connections, but with this. Is that ok?

mikolak, 2020-11-19:
What you've pasted is not an error, but a stringified BleManager.

There are no results, because you're filtering by the device name (`SensorTag`).

Instead of:
```typescript
       if (device.name === 'TI BLE Sensor Tag' || 
            device.name === 'SensorTag') {
            
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();

            // Proceed with connection.
        }
```
put:
```typescript
console.log(device.name)
```

Girum, 2020-11-19:
Yeah, I put this before you write hahaha. It now shows the name. Well, I think now I can test the another functions. Thanks guys!

closed 2020-11-19 by Girum

dariuszseweryn, 2020-11-19:
Hey @Girum 

Great to hear that you are finally able to do your thing. One advice before you will open your next issue on an open source project:

Try to make your research before opening an issue with "it does not work for me" topic. Usually there is quite a substantial number of people using the project already who do not experience issues. Another warning signal should be if your are testing a very basic functionality and it does not seem to work. In other words you should prove (to yourself and to everyone else) that the problem lies in the open source project itself so the maintainers would not spend time for nothing. There are a lot of users and only few people maintaining.

Cheers

Girum, 2020-11-19:
Ok. Thanks for the advice. Since I not used the React native for a time, changes come, and I thinks it can be for that reason. I tested the basic, but forgot the case of this inside the functions in onPress. I tried search many pages and had not found the issue. I used another lib but saw that lib is more complete and powerful, so i tried more.