dotintent/react-native-ble-plx #771: Bug Report

cmmouritsen, 2020-11-19:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

I am expecting to be able to write to a characteristic without errors.
`await this.writeCharacteristic.writeWithResponse(base64Data);`

## Current Behavior

Occasionally I get errors about the characteristic identifier not being found. It appears that there may occasionally be an issue with an extra decimal getting included in the identifier, which causes problems.

`Characteristic 10790551552.0 not found` 

### Steps to Reproduce

This doesn't occur every time, but maybe 30% of the time, especially when I'm writing a lot of data, broken up in chunks.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.1
* Platform: iOS (I haven't tested on Android yet)
* Platform logs (logcat/XCode): Let me know if needed and I can look at what logs I have that would be helpful.
* JS logs: Same
* Contents of the `package.json` file: Same
* Formatted code sample or link to a repository: Let me know if additional examples are needed.

cmmouritsen, 2020-11-20:
I have maybe misdiagnosed this issue. As far as I can tell the characteristic still gets written to, but the error I posted above is logged. The issue is maybe in my code, I'm just not sure why it works fine sometimes and not others. I assume this error is logged just when the native code can't find the characteristic? Is there a property I can check in the javascript to know this will be an issue?

mikolak, 2020-11-20:
Set log level to verbose and run from xcode or android studio to get platform logs. I assume the error happens in the middle of connection, not after reconnecting or something?

cmmouritsen, 2020-11-20:
I can't seem to reproduce now. I'm going to close this and if it comes up again I'll check out the logs and, if needed, log a more accurate issue. Thanks.

closed 2020-11-20 by cmmouritsen