dotintent/react-native-ble-plx #773: Issue with manager.startDeviceScan callback not being fired on iOS when used outside manager.onStateChange callback

qhdcrobey, 2020-11-23:
## Expected Behaviour

I have created a button that sets a state variable, `scanning` to `true`. I am using an effect that has this state variable as a dependancy and will start scanning when it is set to `true`:

```
  useEffect(() => {
    if(scanning) {
      setDevices([]);
      scanAndConnect();
    }
  }, [scanning]);
```

My scanAndConnect() function:

```
const scanAndConnect = function() {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        // Handle error (scanning will be stopped automatically)
        return;
      }

      console.log(device);
    });
  };
```

However I am finding that this is not working (the function gets called however it seems like the listener callback is never fired), and the only way to get the scanning working (on iOS) is to fire my scanAndConnect() function inside the `manager.onStateChange` callback:

```
  const subscription = manager.onStateChange((state) => {
    if (state === 'PoweredOn' && scanning === false) {
      setCanScan(true);
      subscription.remove();
      scanAndConnect(); #the function properly gets called here
    }
  }, true);
```

Please describe the behaviour you are expecting

## Current Behaviour

See above

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

See above

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: ^2.0.1
* Platform: Android/iOS.
* Platform logs (logcat/XCode): No obvious errors or anything to point to an issue
* JS logs: No obvious errors or anything to point to an issue
* Contents of the `package.json` file: 

```
{
  "name": "Boilerplate",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "react-native start",
    "test": "jest",
[... 199 lines cut ...]
                if (a.rssi > b.rssi) {
                  return -1;
                } else if(a.rssi < b.rssi) {
                  return 1;
                }
                return 0;
              }).map(function(device) {
                if(typeof device.device_id !== 'undefined') {
                  return (
                    <Chip key={device.id} style={{marginTop:5}} icon="information" onPress={() => handleDevicePressed(device.asset_id)}>
                      {device.name + ' - ' + device.type.name} - seen { dayjs(device.timestamp).toNow(true) } ago @ {device.rssi}
                    </Chip>
                  );
                }
              }) }
            </View>
          </View>
        </View>
      </View>
    </View>
  );
}

export default AssetScreen;
```

dariuszseweryn, 2020-12-05:
Let me cite the doc of `BleManager`

> /**
>  *
>  * BleManager is an entry point for react-native-ble-plx library. It provides all means to discover and work with
>  * {@link Device} instances. It should be initialized only once with `new` keyword and method
>  * {@link #blemanagerdestroy|destroy()} should be called on its instance when user wants to deallocate all resources.
>  *
>  * In case you want to properly support Background Mode, you should provide `restoreStateIdentifier` and
>  * `restoreStateFunction` in {@link BleManagerOptions}.
>  *
>  * @example
>  * const manager = new BleManager();
>  * // ... work with BLE manager ...
>  * manager.destroy();
>  */

You are using RN hooks. Guess what may be happening in your code and what can be done to prevent it?

qhdcrobey, 2020-12-06:
@dariuszseweryn Thanks for your comment... it did lead me to discover this issue https://github.com/Polidea/react-native-ble-plx/issues/603 where it looks like there are issues with the way I am doing things.. seems like my useEffect may be getting called too early before the manager is ready, also looks like I need to pass the `restoreStateIdentifier` and `restoreStateFunction`... are you able to point me in the right direction?

dariuszseweryn, 2020-12-07:
> It should be initialized only once with new keyword and method
{@link #blemanagerdestroy|destroy()} should be called on its instance when user wants to deallocate all resources

You are creating a new manager with each render in the above mentioned code. Waiting for the correct state is also important.

qhdcrobey, 2020-12-08:
@dariuszseweryn Thank you again. I will close this issue until I implement the changes that you have nudged me towards making. Really appreciate the help. If I am still having troubles I will reopen it / create a new issue.

closed 2020-12-08 by qhdcrobey