dotintent/react-native-ble-plx #774: why android says isConnected befor bonding?

marcosanson, 2020-11-24:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Android 9
Device: OnePlus 3T

my code:

```
console.log("BEFORE CONNECT");
await device.connect();
console.log("AFTER CONNECT", device.isConnected());
```

I expect the second log happens after bonding and code wait till user insert the code. 
Any suggestions?

mikolak, 2020-11-24:
Bond and connection are two separate things. You can be connected and not bonded. If your device supports it, you can discover services and try reading a restricted characteristic, first to trigger bonding, then to make sure the device is bonded. If it doesn't support it, you'll need to make a native module wrapping the call to `createBond() `.

marcosanson, 2020-11-24:
Thank you @mikolak.
I solved chaining these two command right after connect():
 - scan characteristics and services
 - read firmware version from bt device

btw, isBonded() could be a good method to be implemented.

closed 2020-11-24 by marcosanson