dotintent/react-native-ble-plx #779: Could not find rxandroidble-1.7.1.jar (com.polidea.rxandroidble:rxandroidble:1.7.1).

n-maiya, 2020-11-30:
I just added the library to my react native project using npm install --save react-native-ble-plx
My RN version is 0.63.3
On building the app after adding import { BleManager } from 'react-native-ble-plx'; I'm getting this error.

FAILURE: Build failed with an exception.

    What went wrong:
    Execution failed for task ':app:mergeDebugJavaResource'.

    Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
    Could not find rxandroidble-1.7.1.jar (com.polidea.rxandroidble:rxandroidble:1.7.1).
    Searched in the following locations:
    https://jcenter.bintray.com/com/polidea/rxandroidble/rxandroidble/1.7.1/rxandroidble-1.7.1.jar

I tried to make a few changes in build.gradle but nothing seems to work.

dariuszseweryn, 2020-12-04:
Could you share what repositories you have set in your `build.gradle` file?

dariuszseweryn, 2020-12-04:
Closing this. Duplicate of #775

closed 2020-12-04 by dariuszseweryn

angelos3lex, 2022-05-19:
In my case, i had to run a `./gradlew clean` inside the `android` folder, and then this problem got away.