dotintent/react-native-ble-plx #780: Cancelling one monitor via transactionID is buggy.

Tadejo9, 2020-12-03:
I am trying to cancel monitorCharacteristicForDevice using bleManager.cancelTransaction(transactionId).

I wrap monitorCharacteristicForDevice in a promise and resolve that promise with either success or false(on error) string.
When promise resolves I try to cancel transaction with bleManager.cancelTransaction(transactionId) and it gets canceled.
When trying to setup new monitorCharacteristicForDevice, callback is not executed and therefore I cannot get values from second characteristic. 
If i cancel both transaction in the end, flow is ok, but it is not expected behaviour. 

```
await this.listenForChar1(deviceId) // This returns ok result
this.bleManager.cancelTransaction(TRANSACTION1) // This gets canceled
await this.listenForChar2(deviceId) // It enters method but subscription is not setup.
this.bleManager.cancelTransaction(TRANSACTION2) //Is redundant beacuse transaction gets cancelled in promise due to time taken to execute callback.
```
Developing on macos.
Problem on all devices.
Using the latest version of the library.

dariuszseweryn, 2020-12-04:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/react-native-ble-plx/blob/master/.github/ISSUE_TEMPLATE/bug_report.md)?  
It is extremely important to enter as much info as possible or else I cannot help you.

Best Regards

Tadejo9, 2021-03-15:
I'm closing the issue. Problem was solved by not canceling transactions at all. I canceled them when both where recieved or timer went out.

closed 2021-03-15 by Tadejo9