dotintent/react-native-ble-plx #781: How to work with advertisement/manufacturer data?

andrewhavens, 2020-12-08:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

I'm trying to figure out how to work with the advertisement/manufacturer data. I'm new to working with Bluetooth LE. I was reading through the source code, and found that the advertisement data is encoded as base64. I tried to decode it, but I'm not sure what to decode it into (what the encoding is supposed to be) or how to read the data correctly. I'm running this on an Android device.

```js
start_scanning() {
    const uuids = null
    const options = null
    this.manager.startDeviceScan(uuids, options, (error, device) => {
      if (!error) {
        console.log(`Bluetooth device found: (name: ${device.name}, id: ${device.id}), signal strength: ${device.rssi}, advertisement data`, Buffer.from(device.manufacturerData, 'base64').toString('ascii'))
      }
    })
  }
```
Result:
```
[16:16:03] I | ReactNativeJS ▶︎ 'Bluetooth device found: (name: null, id: 61:22:B9:FF:DA:EE), signal strength: -74, advertisement data', 'L\u0000\u0010\u0005F\u001cK#X'
```

dariuszseweryn, 2020-12-08:
Bluetooth is a protocol for transmitting bytes. And that is what is encoded — you decode a base64 string to an array of bytes. How to interpret the bytes is up to who has implemented the higher level protocol.

andrewhavens, 2020-12-08:
Ah okay. That makes more sense. Thanks!

dariuszseweryn, 2020-12-08:
You're welcome.

closed 2020-12-08 by dariuszseweryn