dotintent/react-native-ble-plx #784: Error to enable notifications for notify properties

renatoramossilva, 2020-12-15:
Hello,

i am developing a app to get all measurements from a glucose meter and basically the app is reading 3 set of informations
Service / Characteristic
1808 / 2a18
1808 / 2a24
180A / All characteristics for this service 

For to read the info about the characteristics form the 180a service is easy, because these properties are READ. So, I can get these info using "readCharacteristicForDevice"

In case the service 1808 the characteristics are "NOTIFY", so I need use "monitorCharacteristicForDevice" and write by RACP (0x0101 - for all measurements) and it works for one specific glucose model.

My problem is that for the second one device model I need to enable these characteristics before to read the info. When I try read the info from the second model I received the follow message error:

`BleError: Characteristic 00002a52-0000-1000-8000-00805f9b34fb write failed for device 54:6C:0E:A0:7B:6D and service 00001808-0000-1000-8000-00805f9b34fb`

My question is, how I can enable these characteristics scs using react-native-ble-plx lib?

I am using this function to do it, but it doesn’t nt work yet.

```
enable_notify(device) {
    console.log('enable_notify...');
    this
      .discoverAllServicesAndCharacteristicsForDevice(device.id)
      .then(() => {
        this.servicesForDevice(device.id).then(services => {
          this
            .characteristicsForDevice(
              device.id,
              service,
            )
            .then(() => {
              try {
                var myData = '0x0001';

                const dataMostSignificantByte = (myData >> 8) & 0xff;
                const dataLeastSignificantByte = myData & 0xff;

                const dataByteArrayInLittleEndian = [
                  dataLeastSignificantByte,
                  dataMostSignificantByte,
                ];
                const dataInBase64 = Buffer.from(
                  dataByteArrayInLittleEndian,
                ).toString('base64');

                this
                  .writeCharacteristicWithResponseForDevice(
                    device.id,
                    "00001808-0000-1000-8000-00805f9b34fb",
                    "00002a34-0000-1000-8000-00805f9b34fb",
                    dataInBase64,
                  )
              } catch (err) { console.log('Error (write): ' + err); }
            });
        });
      });
  } 
```
I am writing 0100 because I am using the nrF Logger to understand the message flow and I found it.

D	19:45:31.328	gatt.setCharacteristicNotification(00002a34-0000-1000-8000-00805f9b34fb, true)
V	19:45:31.336	Enabling notifications for 00002a34-0000-1000-8000-00805f9b34fb
D	19:45:31.344	gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	19:45:31.495	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
I	19:45:31.503	Notifications enabled

dariuszseweryn, 2020-12-16:
[wiki](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F)

joaovitorqueiroz, 2021-06-03:
@renatoramossilva Did you manage to take the glucose measurements? I'm having trouble getting the values

yaswanthccbp, 2022-01-27:
@renatoramossilva  hey , we also facing same issue like yours . can you solve the issue? please help to me .

renatoramossilva, 2022-01-27:
> @renatoramossilva Did you manage to take the glucose measurements? I'm having trouble getting the values

Yes! I did! What is your issue?

yaswanthccbp, 2022-01-27:
@renatoramossilva 
we are working on a device by using this we got services and their respective characteristics,among those characteristics we have one characteristic with notify property(isNotifiable:true,isNotifying:false) and a client characteristic configuration descriptor. now we want to make this isNotifying property value to true so that we can get some data stream from the device i refered [this document](https://medium.com/@shahar_avigezer/bluetooth-low-energy-on-android-22bc7310387a) ,in this there is a statement like "to set the notification value, we need to tell the sensor to enables us this notification mode. We will write to the characteristic’s descriptor to set the right value: Notify or Indicate".

in ble-plx there is a method bleManager.writeDescriptorForDevice(), we tried using this but ended up with an error,

so,help me in getting the device data by making isNotifying property value true

keerthi552, 2022-01-27:
@renatoramossilva 
in your code you have used base64 value in writeCharacteristicWithResponseForDevice ,can we give any value or there is a specific base64 value,if it is a specific then how can we get that specific value

jaimebg, 2022-01-31:
@renatoramossilva I'm also having problems receiving notifications, isNotifying is always false and I cannot seem to find any doc on how to activate it, monitorCharacteristicForDevice is not responding anything

diegobonagurio, 2022-06-13:
@renatoramossilva any position about this problem?

diegobonagurio, 2022-06-16:
After numerous attempts, no need to enable notification.

this worked for me:

[image]

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska