dotintent/react-native-ble-plx #785: Error thrown when used with expo-sqlite package

kieranmaine, 2020-12-16:
## Question

I've successfully used react-native-ble-plx with past projects, but with my current project I receive the following error when calling `npx react-native run-android`:

```
* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > File 'com.android.builder.files.ZipCentralDirectory@35c33e4d' was deleted, but previous version not found in cache
```

It appears that this is caused by the `expo-sqlite` package being installed. When I remove this package everything works as expected.

Any ideas what's going on? Is this a bug or do I need to modify my setup?

dariuszseweryn, 2020-12-16:
AFAIK this library does not use anything connected to sqlite. Never encountered such problem.

kieranmaine, 2020-12-16:
I took another look at the problem this morning. By reinstalling all my packages the error has now disappeared! Apologies for the red herring.

closed 2020-12-16 by kieranmaine