dotintent/react-native-ble-plx #786: Bug Report

karkolia, 2020-12-17:
## Expected Behavior

When I call the method **BleManager.cancelDeviceConnection(deviceId)**, the connection to the device should end and I should be able to scan and find the same device again.

## Current Behavior

On iOS only, every time I call **BleManager.cancelDeviceConnection(deviceId)** method, it seems to succeed but when I scan for devices I can't get to find the same device. When I disable and enable manually the bluetooth the device appears again when scanning and also when I kill the app and launch it again.

### Steps to Reproduce

1. Scan
2. Connect to one device
3. DisconnectFrom device
4. Scan => can't find the same device
5. Kill the app and relaunch => finds the device again

### Context

* Library version: "react-native-ble-plx": "^2.0.1",
* Platform:iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
```
{
  "name": "gxxxxxxxxx",
  "version": "1.0.0",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "ios:device": "react-native run-ios --device 'iPhone de D'",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint .",
    "env:staging": "node scripts/set-environment.js staging",
    "env:prod": "node scripts/set-environment.js production"
  },
  "dependencies": {
    "@react-native-community/async-storage": "^1.11.0",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/toolbar-android": "^0.1.0-rc.2",
    "@react-native-firebase/app": "^10.2.0",
    "@react-native-firebase/crashlytics": "^10.2.0",
    "@react-navigation/bottom-tabs": "^5.7.2",
    "@react-navigation/native": "^5.7.1",
    "@react-navigation/stack": "^5.7.1",
    "axios": "^0.19.2",
    "i18n-js": "^3.7.1",
    "lodash.memoize": "^4.1.2",
    "react": "16.11.0",
    "react-native": "0.62.2",
    "react-native-ble-plx": "^2.0.1",
    "react-native-fs": "^2.16.6",
    "react-native-gesture-handler": "^1.6.1",
    "react-native-inappbrowser-reborn": "^3.4.0",
    "react-native-localize": "^1.4.0",
    "react-native-mail": "^6.0.0",
    "react-native-permissions": "^2.1.5",
    "react-native-reanimated": "^1.9.0",
    "react-native-safe-area-context": "^3.1.1",
    "react-native-screens": "^2.9.0",
    "react-native-svg": "^12.1.0",
    "react-native-svg-transformer": "^0.14.3",
    "react-native-svg-uri": "^1.2.3",
    "react-native-vector-icons": "^7.0.0",
    "rxjs": "^6.6.3"
  },
  "devDependencies": {
    "@babel/core": "^7.10.5",
    "@babel/runtime": "^7.10.5",
    "@react-native-community/eslint-config": "^1.1.0",
    "babel-jest": "^26.1.0",
    "eslint": "^7.5.0",
    "jest": "^26.1.0",
    "metro-react-native-babel-preset": "^0.59.0",
    "prettier": "^2.0.5",
    "react-test-renderer": "16.11.0"
  },
  "jest": {
    "preset": "react-native"
  }
}

```

dariuszseweryn, 2020-12-17:
code? logs? are you sure that your device does not change its mac address?

dquand91, 2021-01-05:
I am facing a similar issue on IOS device only.
It seem to IOS device can not un-pair (forget) the connected bluetooth device programmatically after **BleManager.cancelDeviceConnection(deviceId)** has been called. Then I have to forget the connected bluetooth device manually and everything work well after that.
I found a topic in apple developer forum, [](https://developer.apple.com/forums/thread/4732)

closed 2021-06-11 by stale[bot]