dotintent/react-native-ble-plx #787: Bug Report

idangabbay564, 2020-12-19:
Hey, i am trying to use the library to connect to a bluetooth device via my react native app. currently examining the library.
Basically everything goes fine until I am trying to actually connect to a device.
accessing device bluetooth state, scaning for devices (in here also there are few problems I believe - can get no props other that MTU, DeviceId, rssi - no name, no UUID's no some of them doesn't contain manufacturer data)

Notice: that code is currently tested in android only

## Expected Behavior

the device.connect() / bleManager.connectToDevice() functions will allow connection to a selected device, and later discovering services/characteristics, write/reading data etc.

## Current Behavior
cant connect to any device - 
the device.connect() / bleManager.connectToDevice() functions just doesn't working - getting "device is disconnected" after trying to connect - every device that I tried is doing the same thing - no connection was successfull.

### Context

* Library version: 2.0.2
* Platform: Android
* Contents of the `package.json` file:

* Formatted code sample:

const scan = () => { // function that resolves with a device 
        return new Promise((resolve, reject) => {
            bleManager.startDeviceScan(null, {}, (error, device) => {
                if (!error && device["manufacturerData"]) { //just tried some filters-didnt work also when i removed/changed the filters
                    resolve(device)
                } else if (error) {
                    reject(error)
                }
            })
        })
    }

    useEffect( () => {
        const func = async () => {
            const device = await scan() // scanned device is returned in here
            await bleManager.stopDeviceScan() // stop device scanning for connecting to a device
            const connectedDevice = await device.connect() // I stuck here - getting "device is disconnected" after trying to connect
            //from here discovering services and characteristics, reading and writing data etc.
        }
        func()
    }, [])

##NOTICE:
I am pretty new to the Bluetooth, Bluetooth apps and devices, BLE domain.
I've tried to connect to Airpods, to a bluetooth speaker, to a fitness tracker once.
is it suppose to act like that when the communication doesn't suppose to happen in BLE, but in Bluetooth classic ? how can I know which device supporting the usage of the library ?
PLEASE help me cause this is pretty much starting from scratch in this Bluetooth domain

dariuszseweryn, 2020-12-19:
First of all BLE != Bluetooth Classic. Some peripherals need to be in proper mode to accept connections. Almost all advertisement data is optional so you may get no name, no manufacturer data, etc.

I guess you should try with a peripheral which behaviour you know.

idangabbay564, 2020-12-21:
> First of all BLE != Bluetooth Classic. Some peripherals need to be in proper mode to accept connections. Almost all advertisement data is optional so you may get no name, no manufacturer data, etc.
> 
> I guess you should try with a peripheral which behaviour you know.

understood that BLE != Bluetooth Classic. what does that mean ? i cant even connect to devices such as headphones, speakers using the current library ? 
Some peripherals need to be in proper mode to accept connections? which mode ? just set them on pairing mode how complicated is it ? 
I guess you should try with a peripheral which behaviour you know.? what do you mean which behaviour i know ? my whole goal is connecting to a device and learn the behaviour using the services and chareteristics and playing with the device. how can i know the behaviour before connecting to the device ? 

how can i check if a peripheral is supporting a BLE connection ? 

thanks for the help !

dariuszseweryn, 2020-12-21:
> BLE != Bluetooth Classic, what does that mean ?

That this library is for BLE only and there is more than just BLE in the Bluetooth Specification. Up till now music streaming is done via classic flavour of the protocol.

> Some peripherals need to be in proper mode to accept connections? which mode ? just set them on pairing mode how complicated is it ?
> I guess you should try with a peripheral which behaviour you know.? what do you mean which behaviour i know ? my whole goal is connecting to a device and learn the behaviour using the services and characteristics and playing with the device. how can i know the behaviour before connecting to the device ?

Consult the technical specification of the devices you try to connect. Typically commercial devices have their own custom procedure of connecting / authorising.

closed 2021-06-07 by dariuszseweryn