dotintent/react-native-ble-plx #790: Bug Report

hamzadahmani, 2020-12-22:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [ x] I checked the documentation and found no answer
- [ x] I checked to make sure that this issue has not already been filed

## Expected Behavior

i want to catch error of monitor (notify characteristics) when i remove the notify its work only on android 

## Current Behavior

i don't receive error event

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.2
* Platform: iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
**const notify = characteristics.monitor((error, result) => {
      if (error) {
        //not work in ios
        console.log('error', error);
      } else {
....       
}
```

dariuszseweryn, 2020-12-22:
Hello,
Could you provide more specific information of what you experience on Android and what on iOS?

hamzadahmani, 2020-12-22:
hello 
in android for example after remove notify I have in the error of callback operation cancelled on ios I don't receive
I want to be sure this notify was closed

dariuszseweryn, 2020-12-22:
How do you cancel the notification?

hamzadahmani, 2020-12-22:
i use the remove function 
notify.remove()

dariuszseweryn, 2020-12-22:
Could you check what will happen if you would use `transactionId`?

hamzadahmani, 2020-12-22:
I try this but the same thing in both ways the notify was closed but I don't receive the error event

dariuszseweryn, 2020-12-22:
In both situations (subscription cancel and transaction cancel) Android does emit error and iOS does not. Correct?

hamzadahmani, 2020-12-22:
yes

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska